espressif/vscode-esp-idf-extension #1101: Setup Wizard within Docker container fails.

julianiaccopucci, 2023-12-21:
### OS

MacOS

### Operating System version

Ventura 13.6.2

### Visual Studio Code version

1.85.1

### ESP-IDF version

3.16

### Python version

3

### Doctor command output

Produces no output.

### Description

When running the docker container in VSCode, the setup Wizard starts, but it's unable to install the tools.

So, the image works fine from my computer, but the docker image opened in VSCode can't get to set up correctly.

How to reproduce:

1) Create a new project from template from VSCode
2) Open the project and reopen in container
3)Open the container
4) Setup Wizard pops up:
[image]

5) Select Advanced:
[image]

6) Use existing shows:

[image]

7) After saving all paths I get:

[image]

and the Wizard goes back to the beginning.

Checking the container in Docker, the file /opt/esp/esp_idf.json doesn't exist. 

### Debug Message

```plain
n/a
```

### Other Steps to Reproduce

But I can run 
`docker run --rm -v $PWD:/project -w /project -u $UID -e HOME=/tmp espressif/idf idf.py build`
without a problem in my computer.

I've also setup everything in my local computer.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-12-22:
The thing is, you are using the python executable from the virtual env instead of the system Python and the setup wizard is trying to create a virtual env with it. If you are using the docker container from our extension template you don't need to use the setup wizard, the extension should be already configured from .vscode/.devcontainer.json

julianiaccopucci, 2023-12-22:
Thanks.

I had tried building unsuccessfully before, but I've been able to do it by cleaning the build and running the build from the Terminal window in VSCode (Container). My problem was that the build folder had files from the build outside the container. In any case, I still can't build from the build button. Any ideas will be appreciated.

✅ Open the Terminal window in VSCode (Container) and run: idf.py build. Or if I had built outside the container, then I need to do idf.py clean && idf.py build 
❌ In VSCdoe (Container): ESP-IDF Build your project. Cleaning also doesn't solve any issues
❌ In VSCdoe (Container): Click the little build button at the bottom of the window. Cleaning also doesn't solve any issues

For both I get the same error:

[image]

After this I will be updating the docs regarding the Wizard. Current docs suggest to complete it if required - so when appears I consider it was required - as well as the clean builds tip in case anyone stumbles upon it.

brianignacio5, 2023-12-22:
For some reason you are using settings from host computer instead of docker settings. Is your project's/.vscode/settings.json has a `idf.pythonBinPath` defined ? The doctor command output will copy the output to your clipboard.

julianiaccopucci, 2023-12-22:
I don't have one defined. 
Where do I find the doctor command output? When I run 'ESP-IDF Doctor Command' the Output window and Terminal window are clean.

brianignacio5, 2023-12-22:
After you run the command you can paste it in any editor. We fixed this in latest master to open editor with output later.

Somewhere in your setup the extension is using `idf.pythonBinPath: "/Users/Julian/.espressif/python_env/idf5.3_py3.11_env/bin/python"`

julianiaccopucci, 2023-12-22:
I see. This is the output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux arm64 6.5.11-linuxkit 
System environment variable IDF_PYTHON_ENV_PATH 
 /opt/esp/python_env/idf5.3_py3.10_env 
System environment variable PATH 
 /opt/esp/idf/components/espcoredump:/opt/esp/idf/components/partition_table:/opt/esp/idf/components/app_update:/opt/esp/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/opt/esp/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/opt/esp/tools/cmake/3.24.0/bin:/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/opt/esp/tools/qemu-xtensa/esp_develop_8.1.3_20231206/qemu/bin:/opt/esp/tools/qemu-riscv32/esp_develop_8.1.3_20231206/qemu/bin:/opt/esp/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/opt/esp/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/opt/esp/tools/cmake/3.24.0/bin:/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/opt/esp/tools/qemu-xtensa/esp_develop_8.1.3_20231206/qemu/bin:/opt/esp/tools/qemu-riscv32/esp_develop_8.1.3_20231206/qemu/bin:/opt/esp/python_env/idf5.3_py3.10_env/bin:/opt/esp/idf/tools:/opt/qemu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.85.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.6.5 
Workspace folder /Users/julian/Downloads/template-app 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /opt/esp/idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/opt/esp/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/opt/esp/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: /opt/esp/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) /Users/julian/.espressif/python_env/idf5.3_py3.11_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) /opt/esp
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin: true
Access to /opt/esp/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin: true
Access to /opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin: true
Access to /opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin: true
Access to /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: true
Access to /opt/esp/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin: false
Spaces in /opt/esp/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin: false
Spaces in /opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin: false
Spaces in /opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin: false
Spaces in /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: false
Spaces in /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: false
Spaces in /opt/esp/tools/esp-rom-elfs/20230320: false
Spaces in Virtual env Python Path (idf.pythonBinPath) false
[... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------

brianignacio5, 2023-12-22:
A quick fix will be to set in your .vscode/settings.json: `"idf.pythonBinPath": "/opt/esp/python_env/idf5.3_py3.10_env"`

How are you running vscode in docker container ? What about `.vscode/.devcontainer.json` ? any idf.* setting there ?

julianiaccopucci, 2023-12-22:
I found a mismatch of pythonBinPath. The .devcontainer references the wrong path in the docker container.

Updating the .devcontainer to match the path under the Docker container fixes the issue and I can build now.

[image]

[image]

brianignacio5, 2024-01-15:
Because the docker file image is using espressif/idf master which gets updated frequently (idf version changes) but the template was not updated in a while. We will drop this setting in the future but in the meantime will update to latest version in template.

closed 2024-01-25 by brianignacio5

travisp, 2024-11-07:
I just ran into this — the docker file had 5.5 but the devcontainer had 5.4. Could this get added to the documentation for troubleshooting?

brianignacio5, 2024-11-08:
We removed the `idf.pythonBinPath` from extension in current master docker files template (devcontainer.json) so it is computed from current IDF_PATH and IDF_TOOLS_PATH settings. This means it will not be necessary to update it and we also remove the settings from the documentation.

This will be available in next release v1.9.0

PR #1109 (2024-01-25): update dockerfile devcontainer
## Description

Update Dockerfile and devcontainer.json to latest IDF and python version.

Add Python Lib directory to Path in python env install.

Fixes #1100
Fixes #1101
Fixes #1106

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "Configure ESP-IDF extension" command
2. Execute setup action
3. Observe results. Python libraries used in setup should come from embed Python.

- [... cut]