espressif/vscode-esp-idf-extension #1106: Cannot install ESP-IDF VSCode extension; issue with internal Python installation and path variables -> DLL's failed to load

timf34, 2024-01-07:
### OS

Windows

### Operating System version

Windows 11 

### ESP-IDF version

5.1.2

### Python version

3.11.2 

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.8\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.8\libnvvp;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Git\cmd;C:\ProgramData\chocolatey\bin;C:\Program Files\Amazon\AWSCLIV2\;C:\Program Files (x86)\DVC (Data Version Control);C:\Program Files\Go\bin;C:\Program Files\PuTTY\;C:\gstreamer\1.0\mingw_x86_64;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\nodejs\;C:\gstreamer\1.0\msvc_x86_64\lib\gstreamer-1.0;C:\gstreamer\1.0\msvc_x86_64\bin;C:\Program Files\CMake\bin;C:\Program Files\7-Zip;C:\Program Files\Calibre2\;C:\Windows\System32;C:\Program Files\Docker\Docker\resources\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\Scripts;C:\Program Files\NVIDIA Corporation\Nsight Compute 2022.3.0\;C:\Program Files\OpenSSL-Win64\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\Scripts;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311;C:\Program Files\MySQL\MySQL Shell 8.0\bin\;C:<HOMEPATH>\.cargo\bin;C:<HOMEPATH>\AppData\Local\atom\bin;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files (x86)\chromedriver_win32;C:\ProgramData\timf3\GitHubDesktop\bin;C:\Program Files (x86)\Microsoft\Edge\Application;C:<HOMEPATH>\go\bin;C:\gstreamer\1.0\msvc_x86_64\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Roaming\npm;C:\ffmpeg\bin;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code Insiders\bin;C:\gstreamer\1.0\msvc_x86_64\lib\gstreamer-1.0;C:\Program Files (x86)\Nmap;C:<HOMEPATH>\AppData\Local\Pandoc\;C:\MinGW\bin;C:\Program Files\Java\jdk-21\bin;C:<HOMEPATH>\.dotnet\tools; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.85.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.5 
Workspace folder c:<HOMEPATH>\WebstormProjects\timf34.github.io 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) ${env:IDF_PATH}
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars)
Virtual env Python Path (idf.pythonBinPath) C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) 
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.39.2.windows.1
ESP-IDF version x.x
Python version Not found
Python's pip version Not found
-------------------------------------------------- Project configuration settings ----------------------------------------------------------
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
[... 61 lines cut ...]
    import _ssl             # if we can't import it, let the error propagate
    ^^^^^^^^^^^
ImportError: DLL load failed while importing _ssl: The specified procedure could not be found.
Command failed: "C:\Users\timf3\.espressif\tools\idf-python\3.11.2\python.exe" "C:\Users\timf3\esp\esp-idf\tools\idf_tools.py" install-python-env
Traceback (most recent call last):
  File "C:\Users\timf3\esp\esp-idf\tools\idf_tools.py", line 43, in <module>
    import ssl
  File "C:\Users\timf3\AppData\Local\Programs\Python\Python311\Lib\ssl.py", line 100, in <module>
    import _ssl             # if we can't import it, let the error propagate
    ^^^^^^^^^^^
ImportError: DLL load failed while importing _ssl: The specified procedure could not be found.
Command failed: "C:\Users\timf3\.espressif\tools\idf-python\3.11.2\python.exe" "C:\Users\timf3\esp\esp-idf\tools\idf_tools.py" install-python-env
Traceback (most recent call last):
  File "C:\Users\timf3\esp\esp-idf\tools\idf_tools.py", line 43, in <module>
    import ssl
  File "C:\Users\timf3\AppData\Local\Programs\Python\Python311\Lib\ssl.py", line 100, in <module>
    import _ssl             # if we can't import it, let the error propagate
    ^^^^^^^^^^^
ImportError: DLL load failed while importing _ssl: The specified procedure could not be found.
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-01-08:
Seems to be an error with your libraries in `C:\Users\timf3\AppData\Local\Programs\Python\Python311\Lib\ssl.py` not existing. 

Thanks for pointing it out. Seems that extension setup is trying to find these libraries in your env PATH instead of using those from idv-python. will fix asap

timf34, 2024-01-10:
@brianignacio5 I installed the `[esp-idf-extension-open.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1153724729.zip)` from your Pull Request and installed it into my VSCode (and confirmed that your code changes were contained in it). However I am still unfortunately getting the same error that I originally got during installation, choosing Express mode and then the 5.1.2 version.  

The attenuated terminal output during install is as follows:
```
(installing other packages successfully)
Installing package ESP ROM ELFs
Installing tar.gz package ESP ROM ELFs

Traceback (most recent call last):
  File "C:\Users\timf3\esp\esp-idf\tools\idf_tools.py", line 43, in <module>
    import ssl
  File "C:\Users\timf3\AppData\Local\Programs\Python\Python311\Lib\ssl.py", line 100, in <module>
    import _ssl             # if we can't import it, let the error propagate
    ^^^^^^^^^^^
ImportError: DLL load failed while importing _ssl: The specified procedure could not be found.
Command failed: "C:\Users\timf3\.espressif\tools\idf-python\3.11.2\python.exe" "C:\Users\timf3\esp\esp-idf\tools\idf_tools.py" install-python-env
Traceback (most recent call last):
  File "C:\Users\timf3\esp\esp-idf\tools\idf_tools.py", line 43, in <module>
    import ssl
  File "C:\Users\timf3\AppData\Local\Programs\Python\Python311\Lib\ssl.py", line 100, in <module>
    import _ssl             # if we can't import it, let the error propagate
    ^^^^^^^^^^^
ImportError: DLL load failed while importing _ssl: The specified procedure could not be found.
Command failed: "C:\Users\timf3\.espressif\tools\idf-python\3.11.2\python.exe" "C:\Users\timf3\esp\esp-idf\tools\idf_tools.py" install-python-env
Traceback (most recent call last):
  File "C:\Users\timf3\esp\esp-idf\tools\idf_tools.py", line 43, in <module>
    import ssl
  File "C:\Users\timf3\AppData\Local\Programs\Python\Python311\Lib\ssl.py", line 100, in <module>
    import _ssl             # if we can't import it, let the error propagate
    ^^^^^^^^^^^
ImportError: DLL load failed while importing _ssl: The specified procedure could not be found.
```

The Doctor command is practically identical too:
```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621
System environment variable IDF_PYTHON_ENV_PATH
undefined
System environment variable PATH
C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.8\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.8\libnvvp;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;C:\Program Files\dotnet;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Git\cmd;C:\ProgramData\chocolatey\bin;C:\Program Files\Amazon\AWSCLIV2;C:\Program Files (x86)\DVC (Data Version Control);C:\Program Files\Go\bin;C:\Program Files\PuTTY;C:\gstreamer\1.0\mingw_x86_64;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C:\Program Files\nodejs;C:\gstreamer\1.0\msvc_x86_64\lib\gstreamer-1.0;C:\gstreamer\1.0\msvc_x86_64\bin;C:\Program Files\CMake\bin;C:\Program Files\7-Zip;C:\Program Files\Calibre2;C:\Windows\System32;C:\Program Files\Docker\Docker\resources\bin;C:\AppData\Local\Programs\Python\Python311;C:\AppData\Local\Programs\Python\Python311\Scripts;C:\Program Files\NVIDIA Corporation\Nsight Compute 2022.3.0;C:\Program Files\OpenSSL-Win64\bin;C:\AppData\Local\Programs\Python\Python311\Scripts;C:\AppData\Local\Programs\Python\Python311;C:\Program Files\MySQL\MySQL Shell 8.0\bin;C:.cargo\bin;C:\AppData\Local\atom\bin;C:\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files (x86)\chromedriver_win32;C:\ProgramData\timf3\GitHubDesktop\bin;C:\Program Files (x86)\Microsoft\Edge\Application;C:\go\bin;C:\gstreamer\1.0\msvc_x86_64\bin;C:\AppData\Local\Microsoft\WindowsApps;C:\AppData\Roaming\npm;C:\ffmpeg\bin;C:\AppData\Local\Programs\Microsoft VS Code Insiders\bin;C:\gstreamer\1.0\msvc_x86_64\lib\gstreamer-1.0;C:\Program Files (x86)\Nmap;C:\AppData\Local\Pandoc;C:\MinGW\bin;C:\Program Files\Java\jdk-21\bin;C:.dotnet\tools;
System environment variable PYTHON
undefined
Visual Studio Code version 1.85.1
Visual Studio Code language en
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe
ESP-IDF Extension version 1.6.5
Workspace folder c:\WebstormProjects\timf34.github.io
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) ${env:IDF_PATH}
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths)
Custom extra vars (idf.customExtraVars)
Virtual env Python Path (idf.pythonBinPath) C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath)
Git Path (idf.gitPath) C:.espressif\tools\idf-git\2.39.2\cmd\git.exe
[... cut ...]
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
```

brianignacio5, 2024-01-11:
The main issue is that vscode always post append paths to environment PATH variables. Since your local environment keeps appearing first then the error persists. I haven't found a proper way to pre append to PATH but I'll try something now.

brianignacio5, 2024-01-16:
I updated the PR to test updating env variable PATH for Python setup. Could you try again see if the DLL is now using the correct path?

PR #1109 (2024-01-25): update dockerfile devcontainer
## Description

Update Dockerfile and devcontainer.json to latest IDF and python version.

Add Python Lib directory to Path in python env install.

Fixes #1100
Fixes #1101
Fixes #1106

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "Configure ESP-IDF extension" command
2. Execute setup action
3. Observe results. Python libraries used in setup should come from embed Python.

- [... cut]
closed 2024-01-25 by brianignacio5