espressif/vscode-esp-idf-extension #1107: Fresh ESP-IDF installation cannot find cmake and compiler path contains "undefined"

gkovalechyn, 2024-01-08:
### OS

Windows

### Operating System version

windows 10

### Visual Studio Code version

latest

### ESP-IDF version

5.1.2

### Python version

N/A - 3.11 installed by the extension in tools path

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 D:\Programs\FireDaemon OpenSSL 3\bin\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;D:\Programs\Git\cmd;D:\Programs\nodejs\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\dotnet\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;D:\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Roaming\npm;C:<HOMEPATH>\.dotnet\tools 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.85.1 
Visual Studio Code language en 
Visual Studio Code shell C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps\Microsoft.PowerShell_8wekyb3d8bbwe\pwsh.exe 
ESP-IDF Extension version 1.6.5 
Workspace folder f:\Projects\ESP32\sample_project 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) D:\Tools\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) D:\Tools\.esp-idf\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;D:\Tools\.esp-idf\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;D:\Tools\.esp-idf\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;D:\Tools\.esp-idf\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;D:\Tools\.esp-idf\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;D:\Tools\.esp-idf\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;D:\Tools\.esp-idf\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;D:\Tools\.esp-idf\tools\cmake\3.24.0\bin;D:\Tools\.esp-idf\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;D:\Tools\.esp-idf\tools\ninja\1.10.2;D:\Tools\.esp-idf\tools\idf-exe\1.0.3;D:\Tools\.esp-idf\tools\ccache\4.8\ccache-4.8-windows-x86_64;D:\Tools\.esp-idf\tools\dfu-util\0.11\dfu-util-0.11-win64;D:\Tools\.esp-idf\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: D:\Tools\.esp-idf\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: D:\Tools\.esp-idf\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) D:\Tools\.esp-idf\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) D:\Tools\.esp-idf
Git Path (idf.gitPath) \Git\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to D:\Tools\.esp-idf\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to D:\Tools\.esp-idf\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to D:\Tools\.esp-idf\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to D:\Tools\.esp-idf\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to D:\Tools\.esp-idf\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to D:\Tools\.esp-idf\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to D:\Tools\.esp-idf\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to D:\Tools\.esp-idf\tools\cmake\3.24.0\bin: true
Access to D:\Tools\.esp-idf\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to D:\Tools\.esp-idf\tools\ninja\1.10.2: true
Access to D:\Tools\.esp-idf\tools\idf-exe\1.0.3: true
Access to D:\Tools\.esp-idf\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to D:\Tools\.esp-idf\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to D:\Tools\.esp-idf\tools\esp-rom-elfs\20230320: true
[... 54 lines cut ...]
I decided then to try to reinstall the extension. I uninstalled the extension, cleaned up the ESP-IDF tools and base directory and reinstalled it again in the same place. But now, instead of trying to compile, almost all commands throw errors that some tool or another is missing, most of the time being cmake.

I tried creating a new project and noticed that the compiler path was broken:
[image]

If I try to build my old project which has this configuration:
[image]
The ESP-IDF output window just says `INFO: Could not find files for the given pattern(s).`

When I try to run the SDK config I again run into the cmake not found issue:
[image]

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

Yesterday I had no issues with the IDF installation and the extension, but after a restart the extension suddenly started having issues with multiple ESP, RTOS and my own code header and source files. After a reinstall, now no tools installed by the extension are found anymore.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-01-08:
For some reason the git path is broken. See `\Git\cmd\git.exe` in doctor command. Check your settings.json for any unwanted settings update.

gkovalechyn, 2024-01-08:
I tried again with a clearer head and it seems that those were my issues.

The "undefined" in the compiler path is because I had missed not setting the device target before trying to compile:
[image]
After setting it, it now has the correct compiler path.

The error I was having with the multiple error messaged inside both the ESP and freeRTOS includes was an error all the way at the start of the error logs where I had missed an include for `freertos/FreeRTOS.h` which needed to be included before `freertos/event_groups.h`. I had not seen this because the error message overflowed the terminal and I couldn't see it.

I have not been able to replicate the `cmake` and `INFO: Could not find files for the given pattern(s).` errors again after a restart. It could have been related to the previous issues but I am not completely sure. 

I'll close this issue, but in the end the error was on my side.

closed 2024-01-08 by gkovalechyn