espressif/vscode-esp-idf-extension #1108: The ESP32-H2 timed out during internal JTAG debugging

Sail-211010, 2024-01-08:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.85.1 

### ESP-IDF version

5.1.1

### Python version

3.11.2

### Doctor command output

[image]

### Description

There was a problem while debugging JTAG. Previously, the same process could be used to Debug, but now it is not possible
Here's my launch.json
{
 "version": "0.2.0",
 "configurations": [
   {
     "name": "GDB",
     "type": "cppdbg",
     "request": "launch",
     "MIMode": "gdb",
     "miDebuggerPath": "${command:espIdf.getXtensaGdb}",
     "program": "${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
     "windows": {
       "program": "${workspaceFolder}\\build\\${command:espIdf.getProjectName}.elf"
     },
     "cwd": "${workspaceFolder}",
     "environment": [{ "name": "PATH", "value": "${config:idf.customExtraPaths}" }],
     "setupCommands": [
       { "text": "target remote :3333" },
       { "text": "set remote hardware-watchpoint-limit 2"},
       { "text": "mon reset halt" },
       { "text": "thb app_main" },
       { "text": "flushregs" }
     ],
     "externalConsole": false,
     "logging": {
       "engineLogging": true
     }
   }
 ]
}

### Debug Message

```plain
1: (1590) ->(gdb)
1: (1604) <-1003-interpreter-exec console "target remote :3333"
1: (3667) ->~"Remote debugging using :3333\n"
1: (9700) ->~"Ignoring packet error, continuing...\n"
1: (9700) ->&"warning: unrecognized item \"timeout\" in \"qSupported\" response\n"
1: (23786) ->~"Ignoring packet error, continuing...\n"
1: (23786) ->&"Remote replied unexpectedly to 'vMustReplyEmpty': timeout\n"
1: (23786) ->1003^error,msg="Remote replied unexpectedly to 'vMustReplyEmpty': timeout"
1: (23786) ->(gdb)
1: (23787) ->&"\n"
1: (23787) ->^done
1: (23787) ->(gdb)
1: (23790) 1003: elapsed time 22186
1: (23793) Send Event AD7MessageEvent
1: (23796) <--gdb-exit
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-01-10:
I would recommend to replace the `"miDebuggerPath": "${command:espIdf.getXtensaGdb}", to `"miDebuggerPath": "riscv32-esp-elf-gdb"`

brianignacio5, 2024-01-23:
Closing this issue for lack of response. Comment here if you need more help.

closed 2024-01-23 by brianignacio5