espressif/vscode-esp-idf-extension #1113: Installation fails to produce settings.json file

pauljurczak, 2024-01-15:
### OS

Windows

### Operating System version

Ubuntu 23.10

### Visual Studio Code version

1.85.1

### ESP-IDF version

release/v5.2

### Python version

3.11.6

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 6.5.0-14-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.85.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.6.5 
Workspace folder <HOMEPATH>/uc/blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.2_py3.11_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin: false
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin: false
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
[... 165 lines cut ...]
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at /home/paul/esp/esp-idf/tools/cmake/project.cmake:506 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!

 *  The terminal process "/usr/bin/bash '-c', 'cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B /home/paul/uc/blink/build -S /home/paul/uc/blink'" terminated with exit code: 1.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-01-15:
Can you select an IDF Target with `ESP-IDF: Set Espressif device target`command and try building again? Also delete your build directory in your project before selecting IDF target.

Based on the doctor command output your settings are fine. Settings doesn't need to exist in your project .vscode/settings.json. There are User settings (global for all vscode windows), workspace and workspace folder settings as described in https://code.visualstudio.com/docs/getstarted/settings#_settings-precedence

pauljurczak, 2024-01-15:
I deleted `build`, selected my target as `esp32` and `esp-wrover-kit`. I'm getting the same error messages. 

That directory doesn't exist:

[image]

pauljurczak, 2024-01-15:
I removed `~/esp` and `~/.espressif` folders and configured the extension again, this time choosing `v5.1.2 (release version)` instead of `release/v5.2 (release branch)`. No problems with building the blink example. Perhaps, there is a problem with v5.2.

Jefferson-Lopes, 2024-01-21:
I'm having this problem across multiple computers, some on Ubuntu, others on Windows, and another on Zorin OS(fresh install). The problem I'm having is that sometimes when I download my project from a git repository (without the .vscode folder), sometimes the .vscode is correctly created but sometimes is not. So far I haven't recognized any patterns in this problem.

What sometimes works is deleting the /esp and /.espressif folders and installing the same esp-idf version again.

PR #1118 (2024-01-26): Bugfix/tarxz install
## Description

This PR add the following bug fixes

- Fix tar.xz install issues in extension by removing tool version from install command.
- Add idf.notificationMode to `vscode.window.withProgress({},()=>{})` functions to show progress in status bar when idf.notificationMode is set to `All` or `Notification`.
- Add ESP32-P4 to extension keywords in package.json
- Update tasks.json for debug adapter task. 

Fixes #1114
Fixes #1113 
Fixes #1063

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to [... cut]
closed 2024-01-26 by brianignacio5