espressif/vscode-esp-idf-extension #1114: [Feature Request]: Stop using VS Code pop-up notification for builds, flashing project messages.

Seefer, 2024-01-16:
This window pop-up obscures the terminal window output when I click on build or flash etc. I feel the proper place for messages of this nature should be sent to the terminal output windows instead of the GUI-based VS Code notification pop-up.

brianignacio5, 2024-01-17:
With the PR https://github.com/espressif/vscode-esp-idf-extension/pull/1087 you can define how notifications are handle by this extension by using a `ESP-IDF: Select output and notification mode` command which allows to define a `idf.notificationMode` configuration setting to:

1) Show task output and notifications
2) Show only task output
3) Show only notification
4) Show nothing

brianignacio5, 2024-01-19:
Does it happen when there is no errors or only when errors appear @txf- ?

Seefer, 2024-01-19:
It happens when simply building or flashing a project. The pop-up notification window seems redundant considering there's build and flash progress information being shown in the terminal window. I feel such messages are best output to the Terminal window and the notification windows is best used for situations where a problem has occurred, and user response is required.

[image]

txf-, 2024-01-19:
> Does it happen when there is no errors or only when errors appear @txf- ?

It happens on success as well as failure.

brianignacio5, 2024-01-23:
> It happens when simply building or flashing a project. The pop-up notification window seems redundant considering there's build and flash progress information being shown in the terminal window. I [...]

As I mentioned before, you can disable this pop-ups if you want with the mentioned command or configuration setting.

brianignacio5, 2024-01-25:
I've added the PR https://github.com/espressif/vscode-esp-idf-extension/pull/1118 to hide the progress notification based on `idf.notificationMode`. With this PR, Progress notification should be disabled when `idf.notificationMode` is different than `All` or `Notification`

PR #1118 (2024-01-26): Bugfix/tarxz install
## Description

This PR add the following bug fixes

- Fix tar.xz install issues in extension by removing tool version from install command.
- Add idf.notificationMode to `vscode.window.withProgress({},()=>{})` functions to show progress in status bar when idf.notificationMode is set to `All` or `Notification`.
- Add ESP32-P4 to extension keywords in package.json
- Update tasks.json for debug adapter task. 

Fixes #1114
Fixes #1113 
Fixes #1063

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to [... cut]
closed 2024-01-26 by brianignacio5