espressif/vscode-esp-idf-extension #1120: [Feature Request]: How to set or change build path in the extension?

hibfh, 2024-01-29:
I know the command " idf.py -B ${PATH}” can set the path for building. Is there any method to use it in the extension conveniently? if not, hope to add such a function, so I can change the path.

brianignacio5, 2024-01-29:
You can use the `idf.buildPath` for MacOS/Linux or `idf.buildPathWin` for Windows in your project's `.vscode/settings.json` file.

As described in the [settings documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/v1.7.0/docs/SETTINGS.md#esp-idf-specific-settings)

closed 2024-01-29 by brianignacio5

brianignacio5, 2024-02-01:
If you find the extension useful. Please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)