espressif/vscode-esp-idf-extension #1121: Clicking the info icon in menuconfig does not display detailed information

zys91, 2024-01-29:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.85.2

### ESP-IDF version

4.4.6

### Python version

3.8.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\VMware\VMware Workstation\bin\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\dotnet\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files (x86)\NetSarang\Xftp 7\;C:\Program Files (x86)\NetSarang\Xshell 7\;C:\Program Files\Git\cmd;E:\ZYS\工具\ping-tools;E:\ZYS\工具\ESP-TOOLS;C:\Program Files\mingw64\bin\;C:\Program Files\CMake\bin\;D:\Android\Sdk\platform-tools;C:\Program Files\dotnet\;D:\Program Files\MATLAB\R2022b\runtime\win64;D:\Program Files\MATLAB\R2022b\bin;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Program Files\Java\jdk-17\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python38\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python38\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Fiddler;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Bandi\Bandizip;C:\Program Files\Bandizip\;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Roaming\npm 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.85.2 
Visual Studio Code language zh-cn 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.0 
Workspace folder d:\Program\Espressif\esp-idf\examples\wifi\iperf 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) d:\Program\Espressif\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) d:\Program\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;d:\Program\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;d:\Program\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;d:\Program\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;d:\Program\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;d:\Program\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;d:\Program\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;d:\Program\Espressif\tools\cmake\3.23.1\bin;d:\Program\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;d:\Program\Espressif\tools\ninja\1.10.2;d:\Program\Espressif\tools\idf-exe\1.0.3;d:\Program\Espressif\tools\ccache\4.3\ccache-4.3-windows-64;d:\Program\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: d:\Program\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
Virtual env Python Path (idf.pythonBinPath) d:\Program\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM6
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-ftdi.cfg
ESP-IDF Tools Path (idf.toolsPath) d:\Program\Espressif
Git Path (idf.gitPath) d:\Program\Espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to d:\Program\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to d:\Program\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to d:\Program\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin: true
Access to d:\Program\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin: true
Access to d:\Program\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin: true
Access to d:\Program\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin: true
Access to d:\Program\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to d:\Program\Espressif\tools\cmake\3.23.1\bin: true
Access to d:\Program\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin: true
Access to d:\Program\Espressif\tools\ninja\1.10.2: true
Access to d:\Program\Espressif\tools\idf-exe\1.0.3: true
Access to d:\Program\Espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to d:\Program\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
[... 84 lines cut ...]
websocket-client version: 1.6.4
Werkzeug version: 0.16.1
wheel version: 0.36.2
windows-curses version: 2.3.2
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Python requirements from d:\Program\Espressif\esp-idf\requirements.txt are satisfied.
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Python requirements from c:<HOMEPATH>\.vscode\extensions\espressif.esp-idf-extension-1.7.0\esp_debug_adapter\requirements.txt are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF Doctor Command
--------------------------------------------------------------------------------------------------------------------------------------------

### Description

Open SDK Configuration Editor（menuconfig），click info icon in the upper right corner of the entry does not display detailed information，plugin version 1.6.5 is ok，update to 1.7.0 occurs this bug

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-01-30:
Hi @zys91 thanks for reporting this. Made a fix in this PR https://github.com/espressif/vscode-esp-idf-extension/pull/1123

Please take a look if you can.

zys91, 2024-01-30:
All right, everything seems to be back to normal, thank you

closed 2024-02-07 by brianignacio5

brianignacio5, 2024-02-07:
If you find the extension useful. Please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)

PR #1123 (2024-02-07): fix webview ui issues
## Description

Fix UI issues in SDK Configuration editor and Setup wizard.

Fixes #1121 
Fixes #1126

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF: SDK Configuration Editor" command. Toggle the i icon (information icon). You should be able to see more information.
2. Execute `ESP-IDF: Configure ESP-IDF extension` on Linux. Use setup wizard and in the end should see in the  [... cut]