espressif/vscode-esp-idf-extension #1124: Setting target for ESP32-C3 fails

myles-parfeniuk, 2024-02-01:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.85.1

### ESP-IDF version

5.2.0

### Python version

3.11.5

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\Scripts;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\MATLAB\R2023a\bin;C:\Program Files\Microchip\xc16\v2.10\bin;C:\Program Files\Microchip\xc32\v4.21\bin;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Git\cmd;C:\Program Files\dotnet\;C:\Program Files\PuTTY\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:<HOMEPATH>\.cargo\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\Scripts;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Microchip\xc8\v2.41\bin;C:\ProgramData\myles\GitHubDesktop\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WinGet\Packages\Helix.Helix_Microsoft.Winget.Source_8wekyb3d8bbwe\helix-23.05-x86_64-windows; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.85.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.0 
Workspace folder d:\development\esp\projects\idf_5_2\c3_testing 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\esp\tools\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\esp\tools\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\esp\tools\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:\esp\tools\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:\esp\tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\esp\tools\tools\cmake\3.24.0\bin;C:\esp\tools\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\esp\tools\tools\ninja\1.11.1;C:\esp\tools\tools\idf-exe\1.0.3;C:\esp\tools\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\esp\tools\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\esp\tools\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\esp\tools\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\esp\tools\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:\esp\tools\python_env\idf5.2_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-ftdi.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\esp\tools
Git Path (idf.gitPath) C:\esp\tools\tools\idf-git\2.30.1\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:\esp\tools\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:\esp\tools\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to C:\esp\tools\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin: true
Access to C:\esp\tools\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin: true
Access to C:\esp\tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:\esp\tools\tools\cmake\3.24.0\bin: true
Access to C:\esp\tools\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to C:\esp\tools\tools\ninja\1.11.1: true
Access to C:\esp\tools\tools\idf-exe\1.0.3: true
Access to C:\esp\tools\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:\esp\tools\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to C:\esp\tools\tools\esp-rom-elfs\20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
[... 92 lines cut ...]
--------------------------------------------------------------------------------------------------------------------------------------------

### Description

Cannot set the target correctly despite it working within CLI with esp-idf exported outside of VsCode. 

### Debug Message

```plain
Running idf.py set-target command after setting target through ESP-IDF extension GUI:

![image](https://github.com/espressif/vscode-esp-idf-extension/assets/138414207/91791ba7-9151-4237-b8b1-9507b39256dd)

Resulting error after attempting to flash: 

A fatal error occurred: This chip is ESP32-C3 not ESP32. Wrong --chip argument?
CMake Error at run_serial_tool.cmake:66 (message):

  C:/esp/tools/python_env/idf5.2_py3.11_env/Scripts/python.exe;;C:/Users/myles/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32
  failed.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-02-01:
Can you use the `ESP-IDF: Set Espressif device target` extension command or just click at the target in the status bar

myles-parfeniuk, 2024-02-01:
Yes, I set the target using the extension command before opening this issue, unfortunately it did not correct the problem.

The target appears correctly on the status bar:
[image]

brianignacio5, 2024-02-01:
Could you try closing this terminal and open a new esp-idf terminal ?

myles-parfeniuk, 2024-02-01:
> Could you try closing this terminal and open a new esp-idf terminal ?
Issue resolved, this worked perfectly. Thank you.

closed 2024-02-01 by myles-parfeniuk

brianignacio5, 2024-02-01:
Glad it works!

If you find the extension useful. Please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)