espressif/vscode-esp-idf-extension #1126: The details in IDF menuconfig cannot be opened

Tinyu-Zhao, 2024-02-04:
### OS

MacOS

### Operating System version

12.7.3 

### Visual Studio Code version

1.86.0

### ESP-IDF version

5.1.2

### Python version

3.9.6

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin x64 21.6.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /usr/local/sbin:/usr/local/bin:/home/user/.espressif/tools/tool/v1.0.0/bin:/usr/local/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.86.0 
Visual Studio Code language zh-cn 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.7.0 
Workspace folder <HOMEPATH>/esp/hello-world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.1_py3.9_env/bin/python
Serial port (idf.port) /dev/cu.usbmodem14101
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
[... 114 lines cut ...]
--------------------------------------------------------------------------------------------------------------------------------------------

### Extension

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/14155606/esp_idf_vsc_ext.log)

### Description

Unable to expand details properly when clicking on menuconfig in ESP-IDF, such as in video 1.

https://github.com/espressif/vscode-esp-idf-extension/assets/44396641/afdae3dd-5248-4621-b5c5-0e4955a12a83

Expected behavior, such as video 2.

https://github.com/espressif/vscode-esp-idf-extension/assets/44396641/ced7811a-64f7-43a7-a3e4-69bdcdba062a

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-02-04:
This is a duplicate of #1121

Tinyu-Zhao, 2024-02-04:
Thank you, perfectly resolved.

Tinyu-Zhao, 2024-02-04:
> This is a duplicate of #1121

Thank you, perfectly resolved.

closed 2024-02-04 by Tinyu-Zhao

brianignacio5, 2024-02-07:
If you find the extension useful. Please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)

PR #1123 (2024-02-07): fix webview ui issues
## Description

Fix UI issues in SDK Configuration editor and Setup wizard.

Fixes #1121 
Fixes #1126

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF: SDK Configuration Editor" command. Toggle the i icon (information icon). You should be able to see more information.
2. Execute `ESP-IDF: Configure ESP-IDF extension` on Linux. Use setup wizard and in the end should see in the  [... cut]