espressif/vscode-esp-idf-extension #1128: [Feature Request]: Apptracing logging to host live output

redfast00, 2024-02-06:
We'd like to log over JTAG, since our serial port is already in use by another peripheral. We saw we could use apptracing for this (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html#app-trace-logging-to-host), but looking at the instructions (https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/app_tracing.md), shows that this is not live: you first capture the log, then after it's done, you can decode it.

I would like the logging to be live: immediately after a log statement is executed, I can see it on the computer. I think this might be possible, since VisualGDB does this as well https://visualgdb.com/tutorials/esp32/apptrace/

redfast00, 2024-02-27:
Still relevant

karlp, 2025-01-29:
This does work, "clicking on `start apptrace`" but the apptrace is saved to a file that scrolls and doesn't have any autoscroll, it would be preferably if it showed up as a regular "output" similar to how RTT output shows up with the cortex-debug plugin.

tail -F in a separate window works ~ok though.

karlp, 2025-01-29:
I will note, pressing the "start apptrace" in the vscode UI is _super_ flaky.  it normally results in the application "breaking" as if a breakpoint had been hit.  I've found it more reliable to start apptrace via telnet to the running openocd externally :|

redfast00, 2026-09-21:
I no longer work at the company that this was useful for, but I can see that in general, this would still be desirable.