espressif/vscode-esp-idf-extension #1130: Arduino ESP32 cloning fails in download phase without recovery

edolis, 2024-02-07:
### OS

Windows

### Operating System version

10.0.19045

### Visual Studio Code version

1.86.0

### ESP-IDF version

5.1.2

### Python version

see doctor log

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files\Zulu\zulu-11\bin\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\Windows\System32;C:\Windows;C:\Windows\System32\wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Calibre2\;C:\Windows\System32;C:\Windows;C:\Windows\System32\wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files\Agilent\IO Libraries Suite\bin;C:\Program Files\IVI Foundation\VISA\Win64\agvisa;C;C:\Strawberry\c\bin;C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\Program Files\dotnet\;C:\Program Files\Crucial\Crucial Storage Executive;C:\Program Files (x86)\Common Files\Acronis\SnapAPI\;C:\Program Files (x86)\Common Files\Acronis\VirtualFile\;C:\Program Files (x86)\Common Files\Acronis\VirtualFile64\;C:\Program Files (x86)\Common Files\Acronis\FileProtector\;C:\Program Files (x86)\Common Files\Acronis\FileProtector64\;D:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Program Files (x86)\Gpg4win\..\GnuPG\bin;D:\Program Files\ffmpeg\bin;D:<HOMEPATH>\.espressif\python_env\idf5.1_py3.11_env\Scripts;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\MiKTeX\miktex\bin\x64\;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Roaming\npm;C:\Program Files\Git\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.86.0 
Visual Studio Code language en 
Visual Studio Code shell C:\Program Files\Git\bin\bash.exe 
ESP-IDF Extension version 1.7.0 
Workspace folder d:\MyStuff\Progetti\P014_fermChamber\AUT\P014\blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) D:<HOMEPATH>\esp\v5.1.2\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) D:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;D:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;D:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;D:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;D:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;D:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;D:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;D:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;D:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;D:<HOMEPATH>\.espressif\tools\ninja\1.10.2;D:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;D:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;D:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;D:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: D:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: D:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) D:<HOMEPATH>\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM3
OpenOCD Configs (idf.openOcdConfigs) board/esp32-bridge.cfg
ESP-IDF Tools Path (idf.toolsPath) D:<HOMEPATH>\.espressif
Git Path (idf.gitPath) D:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
Custom terminal executable (idf.customTerminalExecutable) C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to D:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to D:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to D:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to D:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to D:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to D:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to D:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to D:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to D:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to D:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to D:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to D:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
[... 135 lines cut ...]
So the only way forward seems to be deleting the folder `components/arduino` - which is locked by explorer.exe. 
So you have to kill explorer.exe to remove the lock (bad stuff); initiate another explorer, restart VS code, delete the folder, retry... and rinse and repeat.

Please consider fixing with some normal resume capability or the chance of some manual installation.

### Debug Message

```plain
Receiving objects:  18% (9168/48776), 62.40 MiB | 294.00 KiB/s
Receiving objects:  18% (9168/48776), 62.66 MiB | 271.00 KiB/s
Receiving objects:  18% (9168/48776), 62.93 MiB | 277.00 KiB/s
error: 3694 bytes of body are still expected

fetch-pack: unexpected disconnect while reading sideband packet

fatal: early EOF

fatal: fetch-pack: invalid index-pack output

Arduino ESP32 cloning has exit with 128
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

edolis, 2024-02-07:
another kind of error message:

```
Receiving objects:   8% (4278/48776), 20.40 MiB | 374.00 KiB/s
Receiving objects:   8% (4278/48776), 20.76 MiB | 375.00 KiB/s
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly before end of the underlying stream
error: 6787 bytes of body are still expected

fetch-pack: unexpected disconnect while reading sideband packet

fatal: early EOF

fatal: fetch-pack: invalid index-pack output

Arduino ESP32 cloning has exit with 128
```

edolis, 2024-02-07:
and another.
I think I'll give up by now

```
Receiving objects:   8% (4277/48785), 13.54 MiB | 211.00 KiB/s
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 778 bytes of body are still expected

fetch-pack: unexpected disconnect while reading sideband packet

fatal: early EOF

fatal: fetch-pack: invalid index-pack output

Arduino ESP32 cloning has exit with 128Receiving objects:   8% (4277/48785), 13.54 MiB | 211.00 KiB/s
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 778 bytes of body are still expected

fetch-pack: unexpected disconnect while reading sideband packet

fatal: early EOF

fatal: fetch-pack: invalid index-pack output

Arduino ESP32 cloning has exit with 128
```

brianignacio5, 2024-02-08:
This seems to be an issue with your internet connection.

Can you run this on a terminal:

```bash
D:.espressif\tools\idf-git\2.39.2\cmd\git.exe  config --global http.postBuffer 524288000  # Set a larger buffer size
D:.espressif\tools\idf-git\2.39.2\cmd\git.exe  config --global core.compression 0         # Disable compression
```
and try again.

Other suggestions are here https://github.com/orgs/community/discussions/48568

edolis, 2024-02-08:
Thanks Brian - that thread solved my issue: I had to use a VPN, for some reason that traffic is being filtered too much here (Italy).
Now, some further things:
a) can you please provide the rollback commands for those two config commands above (which did not help) - so that I bring git to its previous state?
b) this Arduino ESP thing brought me to download 2.3 GB of data. Is there a way to copy and use it for another project too? I tried creating another sample project and executing the same command seems to initiate the download from scratch - this is clearly not sustainable.
I am writing _seem_ as when I execute the `ESP-IDF:Add Arduino ESP32 as ESP-IDF component` command, this time I get absolutely no message in the output window - but I see the VPN is starting the download activity again. And then fails (using VPN too) - with no message in the output window (just the popup) - see below 
[image]

Can I copy the content to the components/arduino folder and tweak the configuration to achieve the same results as the command? 
thanks

brianignacio5, 2024-02-08:
Yes you can, this command is basically doing steps from this documentation:

https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html

edolis, 2024-02-08:
Thanks Brian for your boundless patience.
For other beginners, I believe the global components folder is the `\esp\v5.1.2\esp-idf\components`  one (in my case, version 5.1.2.)

Given the size/ potential fragility of the download process (size/ no resume option) , I'd recommend that saving in the global components is an option given to the user when executing the command.

closed 2024-02-08 by edolis