espressif/vscode-esp-idf-extension #1132: [Feature Request]: Enable Monitor output to survive restart when in New Window

project-x51, 2024-02-09:
Hi,

I like putting the ESP-IDF monitor into a new window on another monitor rather than having it at the bottom of the main window along with the terminal etc. This way I can see a longer output without having to scroll. I believe this is a new feature in the latest release of Vscode.

To put the monitor into a new window right click on the "ESP-IDF Monitor" title on the right side of the monitor window and select 'Move Terminal into New Window". Then move the window onto another screen.

The problem I encounter is that every time I "build, flash and monitor' my code the monitor window disappears and a new one opens in the main window again. I do the same thing with the build output window, and it doesn't reset every time I build.

This also happens if I just click on the "ESP-IDF Monitor Device" button.

Please change the way the monitor works so its window is not reset every time I "build, flash and monitor', or open it.

Thanks!

brianignacio5, 2024-08-29:
Sorry for late response on this. We had some high priority tasks and couldn't put effort here.

@project-x51 Please review this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1867728839.zip) it should persist the existing IDF Monitor terminal.

project-x51, 2024-08-29:
@brianignacio5 That works nicely. Awesome, worth the wait. Thanks!!!!!

I created another issue to add a pause button to the monitor output. I believe someone replied that it can be paused with a key combination but I'll never remember that.

PR #1289 (2024-09-10): use static monitor terminal
## Description

Please include a summary of the change and which issue is fixed.

Fixes #1132 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF: Monitor". Click on "ESP-IDF: Build, flash and monitor". Monitor terminal should be reused and the same.
2. 

## How has this been tested?

Manual testing of above.

**Test Configuration**:
* ESP-IDF Version: 5.2
* OS (Windows,Li [... cut]
closed 2024-09-10 by brianignacio5