espressif/vscode-esp-idf-extension #1133: Getting error "Something went wrong while getting serial port list"

Louisg0709, 2024-02-09:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.86.1

### ESP-IDF version

5.1.2

### Python version

3.12.1

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Oculus\Support\oculus-runtime;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\LLVM\bin;C:\Program Files\CMake\bin;C:\Program Files\dotnet\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:<HOMEPATH>\.cargo\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Launcher\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:<HOMEPATH>\.dotnet\tools;C:\Program Files\Neovim\bin;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.86.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.0 
Workspace folder d:\proj\HelloESP\hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\v5.1.2\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
[... 155 lines cut ...]
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1396887\n\tat ChildProcess.exithandler (node:child_process:423:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat maybeClose (node:internal/child_process:1091:16)\n\tat Process.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2024-02-09T23:06:01.589Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1396887\n\tat ChildProcess.exithandler (node:child_process:423:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat maybeClose (node:internal/child_process:1091:16)\n\tat Process.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2024-02-09T23:06:02.241Z"}
{"message":"Cannot access filePath: ${env:ADF_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:ADF_PATH}'\n\tat Object.accessSync (node:fs:264:3)\n\tat Object.accessSync (node:electron/js2c/asar_bundle:2:8579)\n\tat ke (\\dist\\extension.js:2:1394006)\n\tat \\dist\\extension.js:2:1739152\n\tat \\dist\\extension.js:2:1740115\n\tat async \\dist\\extension.js:2:1737521","level":"error","timestamp":"2024-02-09T23:07:07.546Z"}
{"message":"Cannot access filePath: ${env:MDF_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:MDF_PATH}'\n\tat Object.accessSync (node:fs:264:3)\n\tat Object.accessSync (node:electron/js2c/asar_bundle:2:8579)\n\tat ke (\\dist\\extension.js:2:1394006)\n\tat \\dist\\extension.js:2:1739330\n\tat \\dist\\extension.js:2:1740115\n\tat async \\dist\\extension.js:2:1737521","level":"error","timestamp":"2024-02-09T23:07:07.547Z"}
{"message":"Cannot access filePath: ${env:ESP_MATTER_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:ESP_MATTER_PATH}'\n\tat Object.accessSync (node:fs:264:3)\n\tat Object.accessSync (node:electron/js2c/asar_bundle:2:8579)\n\tat ke (\\dist\\extension.js:2:1394006)\n\tat \\dist\\extension.js:2:1739422\n\tat \\dist\\extension.js:2:1740115\n\tat async \\dist\\extension.js:2:1737521","level":"error","timestamp":"2024-02-09T23:07:07.547Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1396887\n\tat ChildProcess.exithandler (node:child_process:423:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat maybeClose (node:internal/child_process:1091:16)\n\tat Process.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2024-02-09T23:07:09.321Z"}
[... cut ...]
Can't view/select serial ports. Saw other person with this issue who didn't have the python libraries installed so i check and they were all installed.

### Debug Message

```plain
Something went wrong while getting the serial port list
```

### Other Steps to Reproduce

I tried reinstalling the extension and vscode but that didnt change anything.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Louisg0709, 2024-02-09:
Thank you in advance, any help is appreciated.

radurentea, 2024-02-16:
Hi @Louisg0709,

It seems that your device is not being connected to the machine or is not being recognized. Can you confirm that your device is being seen by your mac?

brianignacio5, 2024-02-18:
```{"user":true,"message":"Something went wrong while getting the serial port list","stack":"Error: No serial ports found\n\tat \dist\extension.js:2:1427458\n\tat process.processTicksAndRejections (node:internal/process/task_queues:95:5)","level":"error","timestamp":"2024-02-09T22:52:02.228Z"}```

colorscheme-xik3, 2024-02-20:
Hi @Louisg0709,

I also have the same problem. My laptop only has typeC ports and im using a dock to convert the usb to typeC but i don't think that can be the problem. Did you find any solution?

colorscheme-xik3, 2024-03-04:
@brianignacio5 sorry for bringing this up again but i still couldn't work around it. The error i get when i run flash is the folllowing:

 *  The terminal process "C:\Users\franc\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe 'C:\Users\franc\esp\v5.1.2\esp-idf\components\esptool_py\esptool\esptool.py', '-p', 'COM3', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '40m', '--flash_size', '2MB', '0x1000', 'bootloader/bootloader.bin', '0x10000', 'bt_spp_acceptor_demo.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 2. 
 
 I can't change COM3 because it says: Something went wrong while getting the serial port list

brianignacio5, 2024-03-07:
This seems to me that your OS can't see available serial ports. 

Did you check that your serial ports are available?

`C:\Users\franc.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:\Users\franc\esp\v5.1.2\esp-idf\components\esptool_py\esptool\esptool.py` can also show you serial devices in your system.

colorscheme-xik3, 2024-03-07:
@brianignacio5 Thank you for the response!. My laptop has only typeC ports. In windows 11 i think the drivers for the COM are not used anymore. At least they are seen as legacy drivers on my device manager. So basically i dont have "COM" ports.

I'm using a USB hub to convert the ESP32 microUSB to typeC. You probably can't help since it's not a problem with espressif but thank you anyway. 

[image]

brianignacio5, 2024-03-08:
I think you should take a look at this document about drivers:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html#configure-usb-drivers

I can't be sure if the issue is your usb hub or what, I've tried usb hub with windows 11 but your specific case not sure. You can always test connection with esptool.py or with this python script (same we use in extension)

```py
import sys

try:
    import serial.tools.list_ports

    print([comport.device for comport in serial.tools.list_ports.comports()])

except ImportError:
    print("Pyserial is not installed for %s. Please, check that IDF dependencies are installed." % sys.executable)
    sys.exit(1)
```

colorscheme-xik3, 2024-03-08:
It's working! Turns out it was the microUSB cable. I tested two they didnt work now i tested a third and it works. I feel kinda stupid. Thank you @brianignacio5 you were amazing anyway.

brianignacio5, 2024-03-12:
Glad it works @colorscheme-xik3 !

Closing this issue due to lack of response from original author. Please comment here if you still have some issues.

If you find the extension useful. Please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)

closed 2024-03-12 by brianignacio5

markusiii, 2024-03-29:
@brianignacio5, I had the same issue. It turned out the hardware on/off switch on my PCB was off... From the error message, I thought there was a software issue. Maybe giving the user a hint with "Could not find any ESP" might be helpful. As you already mentioned, I am not sure if it's your responsibility.

brianignacio5, 2024-04-01:
Thanks for the feedback @markusiii will update the error message to be more descriptive.