espressif/vscode-esp-idf-extension #1134: [Feature Request]: Pause monitor output

project-x51, 2024-02-13:
When I write code that causes the ESP32 to crash due to a panic or similar the monitor output contains a crash dump which I like to read. However typically the code then runs again, crashes and continues to do this in a never ending loop. It very quickly overcomes the buffer for the monitor and then it just scrolls quickly making it difficult to tell what happened. My device has an on/off switch so I just turn it off and then I can read what happened. Not ideal, especially when not all devices have such a switch.

I'd like to be able to pause or stop the monitor using a button on the side of the monitor window. When paused no new data would show until I click the button again to resume monitoring.

As an additional feature it would be cool to also have an auto-pause feature that pauses the log after the crash dump when-ever the device restarts.

Cheers

SolidStateLEDLighting, 2024-02-13:
Open up menuconfig and look for "Panic handler behaviour".   You may change that behavior by adding delay time to the reboot, or stop the reboot after a panic altogether.

I think that will do what you want.

project-x51, 2024-02-13:
> Open up menuconfig and look for "Panic handler behaviour". You may change that behavior by adding delay time before the reboot, or stop the reboot after a panic altogether.
> 
> I think that will do what you want.

Nice, I like this as a work around/solution to my problem. I still think there is value in being able to pause the monitor log but will use this as my solution for now.

Thanks!

SolidStateLEDLighting, 2024-02-13:
You're welcome.

brianignacio5, 2024-03-12:
I'm wondering, do we need a button ?

Here are some keyboard shortcuts for IDF Monitor which allow you to pause for example, (basically stop printing output to terminal): https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/tools/idf-monitor.html#keyboard-shortcuts

We could add a button in the Espressif Tab, or a command that would send a signal to the IDF Monitor if running, if that helps.

project-x51, 2024-03-13:
Hi Brian,

I don't think most VSCode users know you can control the monitor output
using keystrokes as this is typically only done if you launch it from the
command line.

Even though I knew this, I didn't realise you could pause it. So, yes I
think buttons would be useful if this is feasible.

Kind Regards,

Robin Alden

VasylMA, 2025-01-21:
I think it would be good to open the monitor in a separate tab with a short toolbar (clear, copy, pause/resume)

artgodwin, 2025-04-26:
> Hi Brian,
>
> I don't think most VSCode users know you can control the monitor output
> using keystrokes as this is typically only done if you launch it from the
> command line.
>
> Even though I knew this, I [...]

It's broken, though.  ^Y stops for a couple of seconds and then restarts on its own. 
I'd suggest using 'screen' instead. It accepts the more normal ^S/^Q.