espressif/vscode-esp-idf-extension #1135: Plugin spams with messages

chedim, 2024-02-13:
### OS

MacOS

### Operating System version

n/a

### Visual Studio Code version

n/a

### ESP-IDF version

n/a

### Python version

n/a

### Doctor command output

n/a

### Extension

n/a

### Description

The plugin spams with error message notifications every time something goes wrong, which is very annoying as, for example, it requires dismissing notification popups every time a build fails. A user can see that their build failed in the build output window, could you not do this or at least hide the popups after a timeout?

Thank you.

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Segfault1602, 2024-02-16:
A pretty bad example of this: if you unplug your device while the OpenOCD Server is running, OpenOCD gets into a retry loop until the device is found again. Each failed retry generate 2 or 3 notifications every ~1 second until either the device is found again or OpenOCD Server is stopped and at this point you can easily have >20 popup to dismiss.

[image]

radurentea, 2024-02-16:
Hi,

Thanks for the feedback! We'll investigate these specific scenarios to explore alternative approaches that may differ from our existing solutions.

### Present Setup:
To tailor your notification preferences
* navigate to **Command Palette** > **Open Settings (UI),** 
* look for **Idf: Notification Mode**
* choose the option that aligns with your needs. For your situations, "**Output**" might be the most suitable choice.

Alternatively, you can insert the following line into your `settings.json`: "**idf.notificationMode": "Output**"

PR #1157 (2024-03-21): fix openOCD notification error messages
## Description

Remove openOCD to report errors as notifications.

Fixes #1135
Fixes #1148

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "OpenOCD Server" to start openOCD server.
2. Disconnect device without closing openOCD.
3. Observe results.

- Expected behaviour:

Errors should be added in log file and Output channel ESP-IDF but not shown as notification.

- Ex [... cut]
closed 2024-03-09 by github-actions[bot]