espressif/vscode-esp-idf-extension #1136: Zephyr install killed IDF tool chain.... CMake problem - Reinstall did not clear the issue.

SolidStateLEDLighting, 2024-02-13:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.86.1

### ESP-IDF version

5.1.2

### Python version

3.12.2

### Doctor command output

I had been using the ESP-IDF extension with great success for several years.    But now I seem to have a problem that I can't solve.   I typically rely on the IDF extension to install everything for me and I have run multiple IDF version for several projects at with the same install for many months.

BUT -- I was experimenting with installing Zephyr and my CMake environment is now not working.  I have uninstalled and reinstalled everything and searched for the solution -- but I just can't seem to solve this path problem.

Paths are written all over the place!! (in the project, in the system environment variables, etc) Where are CMake paths to be specified and why is the extension failing to do so now?

I have never specified a CMake path before, so I don't know what or where the problem is.

IF YOU HAVE REACHED THIS POINT, here is a pledge to the person who wants to provide the instruction that prevents me from having the problem in the future.   I'm pledging $20USD for the assistance (can you still buy a pizza for the in the US for that price?).   I'm outside the US and my pizzas are only about $10.

**RESULTS: I receive almost the same print out everything time I try anything:**

-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.37.1.windows.1")
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32s3-elf-gcc
CMake Error at C:/Users/Keith/esp/v5.1.2/esp-idf/tools/cmake/project.cmake:448 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32s3-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment    
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:8 (project)

CMake Error at C:/Users/Keith/esp/v5.1.2/esp-idf/tools/cmake/project.cmake:448 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32s3-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:8 (project)

CMake Error at C:/Users/Keith/esp/v5.1.2/esp-idf/tools/cmake/project.cmake:448 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32s3-elf-gcc

[... 66 lines cut ...]
{"user":true,"message":"C:\\Users\\Keith\\esp\\v5.1.2\\esp-idf already exists. Delete it or use another location. (ERROR_EXISTING_ESP_IDF)","stack":"Error: C:\\Users\\Keith\\esp\\v5.1.2\\esp-idf already exists. Delete it or use another location. (ERROR_EXISTING_ESP_IDF)\n\tat \\dist\\extension.js:2:1579584\n\tat async \\dist\\extension.js:2:1579265\n\tat async \\dist\\extension.js:2:1578893","level":"error","timestamp":"2024-02-13T02:05:07.284Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1396887\n\tat ChildProcess.exithandler (node:child_process:423:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat ChildProcess.emit (node:domain:489:12)\n\tat maybeClose (node:internal/child_process:1091:16)\n\tat Process.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2024-02-13T02:05:59.532Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1396887\n\tat ChildProcess.exithandler (node:child_process:423:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat ChildProcess.emit (node:domain:489:12)\n\tat maybeClose (node:internal/child_process:1091:16)\n\tat Process.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2024-02-13T02:06:00.400Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1396887\n\tat ChildProcess.exithandler (node:child_process:423:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat ChildProcess.emit (node:domain:489:12)\n\tat maybeClose (node:internal/child_process:1091:16)\n\tat Process.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2024-02-13T02:06:08.616Z"}
{"message":"non zero exit code 1\r\n\r\nINFO: Could not find files for the given pattern(s).\r\n","stack":"Error: non zero exit code 1\r\n\r\nINFO: Could not find files for the given pattern(s).\r\n\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1393836)\n\tat ChildProcess.emit (node:events:514:28)\n\tat ChildProcess.emit (node:domain:489:12)\n\tat Process.onexit (node:internal/child_process:291:12)","level":"error","timestamp":"2024-02-13T02:07:01.660Z"}
{"message":"Cannot access filePath: xtensa-esp32s3-elf-gcc","stack":"Error: non zero exit code 1\r\n\r\nINFO: Could not find files for the given pattern(s).\r\n\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:2:1393836)\n\tat ChildProcess.emit (node:events:514:28)\n\tat ChildProcess.emit (node:domain:489:12)\n\tat Process.onexit (node:internal/child_process:291:12)","level":"error","timestamp":"2024-02-13T02:07:01.660Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: Task ESP-IDF Build exited with code 1\n\tat \\dist\\extension.js:2:1520389\n\tat g.value (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:160:19731)\n\tat r.y (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:80:1902)\n\tat r.z (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:80:1972)\n\tat r.fire (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:80:2188)\n\tat r.$onDidEndTaskProcess (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:153:52912)","level":"error","timestamp":"2024-02-13T02:07:48.457Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: Task ESP-IDF Build exited with code 1\n\tat \\dist\\extension.js:2:1520389\n\tat g.value (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:160:19731)\n\tat r.y (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:80:1902)\n\tat r.z (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:80:1972)\n\tat r.fire (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:80:2188)\n\tat r.$onDidEndTaskProcess (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:153:52912)","level":"error","timestamp":"2024-02-13T02:09:26.094Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: Task ESP-IDF Compile exited with code 1\n\tat \\dist\\extension.js:2:1520389\n\tat g.value (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:160:19731)\n\tat r.y (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:80:1902)\n\tat r.z (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:80:1972)\n\tat r.fire (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:80:2188)\n\tat r.$onDidEndTaskProcess (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:153:52912)","level":"error","timestamp":"2024-02-13T02:11:22.850Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU [... cut ...] 9:12)\n\tat maybeClose (node:internal/child_process:1091:16)\n\tat Process.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2024-02-13T02:43:42.207Z"}

### Description

Can't compile any app -- and template projects are not reachable by the IDF extension.

### Debug Message

```plain
NA
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2024-02-16:
Hi @SolidStateLEDLighting,

After reinstalling the extension, it's crucial to ensure that you've followed the uninstallation steps outlined in [our documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md#uninstalling-the-plugin). One key step that shouldn't be overlooked is:

>Navigate to your ${VSCODE_EXTENSION_DIR} and thoroughly delete the ESP-IDF plugin folder. For Windows users, this directory is typically located at %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\.

Additionally, I recommend removing any environment variables related to ESP-IDF and any manually added ESP-IDF tools from your Windows system. This cleanup is crucial because the VS Code Extension is designed to manage all necessary configurations on its behalf, including installing ESP-IDF, ESP-IDF tools, and setting up the Python virtual environment. One way to initiate this automated setup process is by executing the "ESP-IDF: Configure ESP-IDF Extension" command within VS Code and go through the configuration wizard.

 You can take a look over the **Extension Activation Self Configuration** part from our [documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md#extension-activation-self-configuration) to understand how the extension does the self-configurations and why having older env variables can affect a fresh reinstall. 

This approach ensures a clean, conflict-free environment for your development work, streamlining the setup process and minimizing potential issues. If you encounter any difficulties or have further questions, don't hesitate to reach out again!

SolidStateLEDLighting, 2024-02-16:
Thank you so much for this in-depth response.   I will give this a go when I get a chance and let you know how it works out.

I think you guys have done some amazing work with the Windows development tools and I dropped PIO a few years ago and have never looked back.

SolidStateLEDLighting, 2024-02-17:
Not working.    I have uninstalled everything as your directed.  I have deleted directories, clear system variables, rebooted, restarted, and reinstalled.

I installed v4.4.6

I have 2 peculiar behaviors.   

A) When I try to set the IDF_PATH through the command pallet -- it keeps pointing to v5.2 (which is a build which I last worked on before installing Zephyr).    I can not force that to move to v4.4.6.   It just won't take this even through v4.4.6 is the only framework on the computer in that path.

B) When I ask to start a new project -- it keeps thinking that I have an ESP_MDF framework available. (in the Extension drop down)    It doesn't register the v4.4.6     Why would it even see an MDF framework when that has been gone?

I have not explicitly touched CMake or Python in this process.  So what am I missing?

brianignacio5, 2024-02-18:
What is your doctor command output from `ESP-IDF: Doctor Command` ?

https://github.com/espressif/vscode-esp-idf-extension?tab=readme-ov-file#troubleshooting

SolidStateLEDLighting, 2024-02-18:
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\dotnet\;C:\Program Files\Amazon\AWSCLIV2\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files\Amazon\AWSSAMCLI_NIGHTLY\bin\;C:\Program Files\Amazon\AWSSAMCLI\bin\;C:<HOMEPATH>\AppData\Local\Programs\Python\Launcher\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:<HOMEPATH>\.dotnet\tools;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:<HOMEPATH>\AppData\Roaming\Python\Python311\Scripts;C:<HOMEPATH>\esp\tools\tools\idf-python\3.11.2\Scripts; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.86.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.0 
Workspace folder c:<HOMEPATH>\Desktop\aws_iot_core 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:/Users/Keith/esp/v4.4.5/esp-idf
ESP-MDF Path (idf.espMdfPath) c:<HOMEPATH>\Desktop\ESP-MDF\esp-mdf-v1.0
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:/Users/Keith/esp/tools/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin;c:/Users/Keith/esp/tools/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin;c:/Users/Keith/esp/tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin;c:/Users/Keith/esp/tools/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin;c:/Users/Keith/esp/tools/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin;c:/Users/Keith/esp/tools/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin;c:/Users/Keith/esp/tools/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin;c:/Users/Keith/esp/tools/tools/cmake/3.23.1/bin;c:/Users/Keith/esp/tools/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin;c:/Users/Keith/esp/tools/tools/ninja/1.10.2;c:/Users/Keith/esp/tools/tools/idf-exe/1.0.3;c:/Users/Keith/esp/tools/tools/ccache/4.3/ccache-4.3-windows-64;c:/Users/Keith/esp/tools/tools/dfu-util/0.9/dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:/Users/Keith/esp/tools/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:<HOMEPATH>\esp\tools\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) c:/Users/Keith/esp/tools/python_env/idf4.4_py3.11_env/Scripts/python.exe
Serial port (idf.port) COM25
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) c:/Users/Keith/esp/tools
Git Path (idf.gitPath) c:/Users/Keith/esp/v4.4.5/tools/tools/idf-git/2.30.1/cmd/git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) true
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:/Users/Keith/esp/tools/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin: true
Access to c:/Users/Keith/esp/tools/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin: true
Access to c:/Users/Keith/esp/tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin: true
Access to c:/Users/Keith/esp/tools/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin: true
Access to c:/Users/Keith/esp/tools/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin: true
Access to c:/Users/Keith/esp/tools/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin: true
Access to c:/Users/Keith/esp/tools/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to c:/Users/Keith/esp/tools/tools/cmake/3.23.1/bin: true
Access to c:/Users/Keith/esp/tools/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin: false
Access to c:/Users/Keith/esp/tools/tools/ninja/1.10.2: true
Access to c:/Users/Keith/esp/tools/tools/idf-exe/1.0.3: true
Access to c:/Users/Keith/esp/tools/tools/ccache/4.3/ccache-4.3-windows-64: true
Access to c:/Users/Keith/esp/tools/tools/dfu-util/0.9/dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
[... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF Doctor Command
--------------------------------------------------------------------------------------------------------------------------------------------

SolidStateLEDLighting, 2024-02-18:
```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\dotnet\;C:\Program Files\Amazon\AWSCLIV2\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files\Amazon\AWSSAMCLI_NIGHTLY\bin\;C:\Program Files\Amazon\AWSSAMCLI\bin\;C:<HOMEPATH>\AppData\Local\Programs\Python\Launcher\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:<HOMEPATH>\.dotnet\tools;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:<HOMEPATH>\AppData\Roaming\Python\Python311\Scripts;C:<HOMEPATH>\esp\tools\tools\idf-python\3.11.2\Scripts; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.86.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.0 
Workspace folder c:<HOMEPATH>\Desktop\aws_iot_core 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:/Users/Keith/esp/v4.4.5/esp-idf
ESP-MDF Path (idf.espMdfPath) c:<HOMEPATH>\Desktop\ESP-MDF\esp-mdf-v1.0
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:/Users/Keith/esp/tools/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin;c:/Users/Keith/esp/tools/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin;c:/Users/Keith/esp/tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin;c:/Users/Keith/esp/tools/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin;c:/Users/Keith/esp/tools/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin;c:/Users/Keith/esp/tools/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin;c:/Users/Keith/esp/tools/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin;c:/Users/Keith/esp/tools/tools/cmake/3.23.1/bin;c:/Users/Keith/esp/tools/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin;c:/Users/Keith/esp/tools/tools/ninja/1.10.2;c:/Users/Keith/esp/tools/tools/idf-exe/1.0.3;c:/Users/Keith/esp/tools/tools/ccache/4.3/ccache-4.3-windows-64;c:/Users/Keith/esp/tools/tools/dfu-util/0.9/dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:/Users/Keith/esp/tools/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:<HOMEPATH>\esp\tools\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) c:/Users/Keith/esp/tools/python_env/idf4.4_py3.11_env/Scripts/python.exe
Serial port (idf.port) COM25
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) c:/Users/Keith/esp/tools
Git Path (idf.gitPath) c:/Users/Keith/esp/v4.4.5/tools/tools/idf-git/2.30.1/cmd/git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) true
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:/Users/Keith/esp/tools/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin: true
Access to c:/Users/Keith/esp/tools/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin: true
Access to c:/Users/Keith/esp/tools/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin: true
Access to c:/Users/Keith/esp/tools/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin: true
Access to c:/Users/Keith/esp/tools/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin: true
Access to c:/Users/Keith/esp/tools/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin: true
Access to c:/Users/Keith/esp/tools/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to c:/Users/Keith/esp/tools/tools/cmake/3.23.1/bin: true
Access to c:/Users/Keith/esp/tools/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin: false
Access to c:/Users/Keith/esp/tools/tools/ninja/1.10.2: true
Access to c:/Users/Keith/esp/tools/tools/idf-exe/1.0.3: true
Access to c:/Users/Keith/esp/tools/tools/ccache/4.3/ccache-4.3-windows-64: true
Access to c:/Users/Keith/esp/tools/tools/dfu-util/0.9/dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
[... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF Doctor Command
--------------------------------------------------------------------------------------------------------------------------------------------

```

SolidStateLEDLighting, 2024-02-18:
I'm not very sharp on just how python should be installed.   I know the extension install a virtual version of it.  Here is how it is identified by command line inside VSC and in Git Bash.

[image]

brianignacio5, 2024-02-18:
Based on your doctor command you should review that these git python virtual environment exist.

For example, does `c:/Users/Keith/esp/v4.4.5/tools/tools/idf-git/2.30.1/cmd/git.exe` exist ?

SolidStateLEDLighting, 2024-02-18:
I have a mis-matching here.   This project was last built with 4.4.5.   My latest install only had 4.4.6 available.

Remember – my current build can't generate a template hello_world project fresh.

My path to this exe is:

c:/Users/Keith/esp/tools/tools/idf-git/2.39.2/cmd/git.exe

SolidStateLEDLighting, 2024-02-18:
Here is what is installed -- and what you see when I try to generate a new project:
[image]
[image]

SolidStateLEDLighting, 2024-02-18:
Do you guys want to give me fresh instructions for a complete reinstall?   I'm not real sharp on Python in terms of what version to choose and how to install it.    I was hoping that the plug-in handled all that.    Same applies for CMake and it's extension.  There are just so many ways to do things now.

SolidStateLEDLighting, 2024-02-18:
When I don't have a project open, this is my Doctor response:

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\dotnet\;C:\Program Files\Amazon\AWSCLIV2\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files\Amazon\AWSSAMCLI_NIGHTLY\bin\;C:\Program Files\Amazon\AWSSAMCLI\bin\;C:<HOMEPATH>\AppData\Local\Programs\Python\Launcher\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:<HOMEPATH>\.dotnet\tools;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:<HOMEPATH>\AppData\Roaming\Python\Python311\Scripts;C:<HOMEPATH>\esp\tools\tools\idf-python\3.11.2\Scripts; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.86.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.0 
Workspace folder No workspace folder is open 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:<HOMEPATH>\esp\v5.2\esp-idf
ESP-MDF Path (idf.espMdfPath) c:<HOMEPATH>\Desktop\ESP-MDF\esp-mdf-v1.0
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:<HOMEPATH>\esp\tools\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;c:<HOMEPATH>\esp\tools\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;c:<HOMEPATH>\esp\tools\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;c:<HOMEPATH>\esp\tools\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;c:<HOMEPATH>\esp\tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;c:<HOMEPATH>\esp\tools\tools\cmake\3.24.0\bin;c:<HOMEPATH>\esp\tools\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;c:<HOMEPATH>\esp\tools\tools\ninja\1.11.1;c:<HOMEPATH>\esp\tools\tools\idf-exe\1.0.3;c:<HOMEPATH>\esp\tools\tools\ccache\4.8\ccache-4.8-windows-x86_64;c:<HOMEPATH>\esp\tools\tools\dfu-util\0.11\dfu-util-0.11-win64;c:<HOMEPATH>\esp\tools\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:<HOMEPATH>\esp\tools\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:<HOMEPATH>\esp\tools\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) c:<HOMEPATH>\esp\tools\python_env\idf5.2_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM25
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) c:<HOMEPATH>\esp\tools
Git Path (idf.gitPath) C:<HOMEPATH>\esp\tools\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) true
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:<HOMEPATH>\esp\tools\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: false
Access to c:<HOMEPATH>\esp\tools\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: false
Access to c:<HOMEPATH>\esp\tools\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin: false
Access to c:<HOMEPATH>\esp\tools\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin: false
Access to c:<HOMEPATH>\esp\tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to c:<HOMEPATH>\esp\tools\tools\cmake\3.24.0\bin: false
Access to c:<HOMEPATH>\esp\tools\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: false
Access to c:<HOMEPATH>\esp\tools\tools\ninja\1.11.1: false
Access to c:<HOMEPATH>\esp\tools\tools\idf-exe\1.0.3: true
Access to c:<HOMEPATH>\esp\tools\tools\ccache\4.8\ccache-4.8-windows-x86_64: false
Access to c:<HOMEPATH>\esp\tools\tools\dfu-util\0.11\dfu-util-0.11-win64: false
Access to c:<HOMEPATH>\esp\tools\tools\esp-rom-elfs\20230320: false
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
[... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF Doctor Command
--------------------------------------------------------------------------------------------------------------------------------------------

```

brianignacio5, 2024-02-18:
The latest doctor command output shows error when trying to find esp-idf version and python version.

Could you try running `c:<HOMEPATH>\esp\tools\python_env\idf5.2_py3.11_env\Scripts\python.exe --version` and that ESP-IDF path is installed correctly.

SolidStateLEDLighting, 2024-02-18:
[image]

brianignacio5, 2024-02-19:
The thing is, the path of your virtual env you used for checking version is different from Doctor Command. 

I will suggest deleting all settings in settings.json that start with `idf.*` in 3 places:

Workspace folder configuration settings in `${workspaceFolder}/.vscode/settings.json`
Workspace configuration settings defined in the workspace's `<name>.code-workspace` file as shown below.
User settings defined in
Windows `%APPDATA%\Code\User\settings.json`

and use a valid setup in your system.

SolidStateLEDLighting, 2024-02-19:
This thing is such a mess that I'm just going to rebuild by drive and start fresh.

Just keep in mind that all this broke loose when I experimented with a Zephyr tools install.  You might want to try a bit of testing to see if you can keep this from happening to others.

Thanks.

closed 2024-02-19 by SolidStateLEDLighting

brianignacio5, 2024-02-20:
Thanks for the feedback, we haven't tested Zephyr with the extension yet but will do. 

Should also add some option to "clean" previous configurations in case of issues.

If you find the extension useful. Please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)