espressif/vscode-esp-idf-extension #1138: Debugging problems in vscode for esp32c6 (devkit) (IDFGH-12122)

vandy, 2024-02-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I've installed `vscode` (Windows 10), cloned `esp-idf` repository (v5.1.2), installed tools to `%userProfile%` with `install.bat`, installed extension for vscode, set everything up. I can build, flash and monitor my firmware.
Wanted to setup debugging via usb-jtag and faced some problems.

1. I've used the default approach with `esp-idf debug adapter` (https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md#use-the-esp-idf-debug-adapter). When I run the Launch (in debug tab), the program stops at `app_main` (which is expected). I can use Step over (or other steps) to proceed. But I can't use Continue (F5) to break at the next breakpoint (to jump from line 9 to line 15 at once). There are errors in the console:
```cmd
2024-02-14 09:54:26,872 - Debug Adapter (main) - ERROR - can't create new thread at interpreter shutdown
Traceback (most recent call last):
  File "%userProfile%\.vscode\extensions\espressif.esp-idf-extension-1.7.0\esp_debug_adapter\debug_adapter\command_processor.py", line 52, in __call__
    on_request(protocol_message)
  File "%userProfile%\.vscode\extensions\espressif.esp-idf-extension-1.7.0\esp_debug_adapter\debug_adapter\command_processor.py", line 208, in on_continue_request
    self.da.resume_exec()
  File "%userProfile%\.vscode\extensions\espressif.esp-idf-extension-1.7.0\esp_debug_adapter\debug_adapter\debug_adapter.py", line 932, in resume_exec
    self.start_target_poller(dbg.TARGET_STATE_STOPPED)
  File "%userProfile%\.vscode\extensions\espressif.esp-idf-extension-1.7.0\esp_debug_adapter\debug_adapter\debug_adapter.py", line 738, in start_target_poller
    self.target_poller.start()
  File "C:\Program Files\Python312\Lib\threading.py", line 992, in start
    _start_new_thread(self._bootstrap, ())
RuntimeError: can't create new thread at interpreter shutdown
```

But if I use command line approach (https://docs.espressif.com/projects/esp-idf/en/stable/esp32c6/api-guides/jtag-debugging/debugging-examples.html#command-line), everything is fine. I can step or continue to jump to the next breakpoint.

Also, using `"type": "cppdbg"` in launch.json with specified gdb binary (from .espressif tools) works ok, I can use Continue to jump between breakpoints.

2. There is the second problem with command line approach. It's said that freertos objects could be debugged using `freertos-gdb` python package. The package is installed. But riscv-...-gdb (from .espressif tools) says that `# Python scripting is not supported in this copy of GDB.` I just cannot import the python package in gdb session.

```cmd
show configuration
This GDB was configured as follows:
   configure --host=x86_64-w64-mingw32 --target=riscv32-esp-elf
             --with-auto-load-dir=$debugdir:$datadir/auto-load
             --with-auto-load-safe-path=$debugdir:$datadir/auto-load
             --with-expat
             --with-gdb-datadir=//share/gdb (relocatable)
             --with-jit-reader-dir=//lib/gdb (relocatable)
             --without-libunwind-ia64
             --without-lzma
             --without-babeltrace
             --without-intel-pt
             --without-mpfr
             --without-xxhash
             --without-python
             --without-python-libdir
             --without-debuginfod
             --without-guile
             --disable-source-highlight
             --with-separate-debug-dir=//lib/debug (relocatable)
```
Should I obtain riscv-...-gdb another way?

Example:
```cpp
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"

static const char *TAG = "MYTAG";

void app_main(void) {
    uint16_t counter = 0;
    ESP_LOGE(TAG, "App main");
    // printf("app_main");
    while (1) {
        counter++;
        ESP_LOGI(TAG, "Counter is %d", counter);
        vTaskDelay(2000 / portTICK_PERIOD_MS);
    }
}
```

vandy, 2024-02-15:
The second problem was solved after update to `esp-idf` v5.2 and running `install.bat` again (to install new tools in another directory using `%IDF_TOOLS_PATH%` env variable). Configuration has changed:
```cmd
show configuration
This GDB was configured as follows:
   configure --host=x86_64-w64-mingw32 --target=riscv32-esp-elf
             --with-auto-load-dir=$debugdir:$datadir/auto-load
             --with-auto-load-safe-path=$debugdir:$datadir/auto-load
             --with-expat
             --with-gdb-datadir=//share/gdb (relocatable)
             --with-jit-reader-dir=//lib/gdb (relocatable)
             --without-libunwind-ia64
             --without-lzma
             --without-babeltrace
             --without-intel-pt
             --without-mpfr
             --without-xxhash
             --without-python
             --with-python-libdir=/opt/python-x86_64-w64-mingw32-3.12.0/lib
             --without-debuginfod
             --without-guile
             --disable-source-highlight
             --with-separate-debug-dir=//lib/debug (relocatable)
```

But to use `freertos` command inside (gdb) session it's required to run `idf.py gdb` (instead of `riscv...gdb -x gdbinit`), or mess with env variables to add python virtual env with installed `freertos_gdb` package (didn't do that).

radurentea, 2024-02-16:
Hi @vandy,

It does look strange, my first thought is that the VS Code extension is misconfigured.

Can you please run the "**Doctor Command**" from VS Code's Command Palette and paste the output from your clipboard here, so we can take a look at VS Code configuration?

vandy, 2024-02-16:
Hi, @radurentea.

```cmd
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Python312\Scripts\;C:\Program Files\Python312\;C:\Program Files\Git\cmd;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\.dotnet\tools;
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.86.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.0 
Workspace folder d:\dev\project\firmware 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) D:\dev\github.com\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:/Users/user/.espressif/idf5.2/tools/ccache/4.8/ccache-4.8-windows-x86_64;C:/Users/user/.espressif/idf5.2/tools/cmake/3.24.0/bin;C:/Users/user/.espressif/idf5.2/tools/dfu-util/0.11/dfu-util-0.11-win64;C:/Users/user/.espressif/idf5.2/tools/esp-rom-elfs/20230320;C:/Users/user/.espressif/idf5.2/tools/idf-exe/1.0.3;C:/Users/user/.espressif/idf5.2/tools/ninja/1.11.1;C:/Users/user/.espressif/idf5.2/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin;C:/Users/user/.espressif/idf5.2/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin;C:/Users/user/.espressif/idf5.2/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin;C:/Users/user/.espressif/idf5.2/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin;C:/Users/user/.espressif/idf5.2/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin;C:/Users/user/.espressif/idf5.2/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\idf5.2\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts
    ESP_ROM_ELF_DIR: C:/Users/user/.espressif/idf5.2/tools/esp-rom-elfs/20230320
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\idf5.2\python_env\idf5.2_py3.12_env\Scripts\python.exe
Serial port (idf.port) COM5
OpenOCD Configs (idf.openOcdConfigs) board/esp32c6-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif\idf5.2
Git Path (idf.gitPath) C:\Program Files\Git\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:/Users/user/.espressif/idf5.2/tools/ccache/4.8/ccache-4.8-windows-x86_64: true
Access to C:/Users/user/.espressif/idf5.2/tools/cmake/3.24.0/bin: true
Access to C:/Users/user/.espressif/idf5.2/tools/dfu-util/0.11/dfu-util-0.11-win64: true
Access to C:/Users/user/.espressif/idf5.2/tools/esp-rom-elfs/20230320: true
Access to C:/Users/user/.espressif/idf5.2/tools/idf-exe/1.0.3: true
Access to C:/Users/user/.espressif/idf5.2/tools/ninja/1.11.1: true
Access to C:/Users/user/.espressif/idf5.2/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: true
Access to C:/Users/user/.espressif/idf5.2/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to C:/Users/user/.espressif/idf5.2/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin: true
Access to C:/Users/user/.espressif/idf5.2/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin: true
Access to C:/Users/user/.espressif/idf5.2/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin: true
Access to C:/Users/user/.espressif/idf5.2/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in C:/Users/user/.espressif/idf5.2/tools/ccache/4.8/ccache-4.8-windows-x86_64: false
Spaces in C:/Users/user/.espressif/idf5.2/tools/cmake/3.24.0/bin: false
Spaces in C:/Users/user/.espressif/idf5.2/tools/dfu-util/0.11/dfu-util-0.11-win64: false
Spaces in C:/Users/user/.espressif/idf5.2/tools/esp-rom-elfs/20230320: false
Spaces in C:/Users/user/.espressif/idf5.2/tools/idf-exe/1.0.3: false
Spaces in C:/Users/user/.espressif/idf5.2/tools/ninja/1.11.1: false
[... cut ...]
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------
```

brianignacio5, 2024-02-18:
Could you share the debug.log file content or the ESP-IDF Output related to debug adapter to see what is the error ?

vandy, 2024-02-18:
[debug.log](https://github.com/espressif/vscode-esp-idf-extension/files/14321964/debug.log)

I've used app_trace basic example from esp-idf with modified app_main:
```c
void app_main(void) {
    ESP_LOGI(TAG, "Waiting for OpenOCD connection");

    // while (!esp_apptrace_host_is_connected(ESP_APPTRACE_DEST_JTAG)) {
    //     vTaskDelay(1);
    // }

    ESP_LOGI(TAG, "Sending example data to the host...");

    for (unsigned int cnt = 1; cnt < 51; ++cnt) {
        char buf[32] = {0};
        snprintf(buf, sizeof(buf), "Apptrace test data[%d]:%d\n", cnt, cnt * cnt);
        esp_err_t res = esp_apptrace_write(ESP_APPTRACE_DEST_JTAG, buf, strlen(buf), ESP_APPTRACE_TMO_INFINITE);
        if (res != ESP_OK) {
            ESP_LOGE(TAG, "Failed to write data to host [0x%x] (%s)", res, esp_err_to_name(res));
        }
        esp_apptrace_flush(ESP_APPTRACE_DEST_JTAG, 1000);
        vTaskDelay(50 / portTICK_PERIOD_MS);
    }

    while (1) {
        ESP_LOGI(TAG, "Loop");
        vTaskDelay(250 / portTICK_PERIOD_MS);
    }

    ESP_LOGI(TAG, "Done!");
}
```

vandy, 2024-03-06:
Hi, @brianignacio5, I've shared the debug.log.

brianignacio5, 2024-03-22:
We are working on migrating to a new debug adapter that is more stable that our current python one in this PR:

https://github.com/espressif/vscode-esp-idf-extension/pull/1131

You are welcome to test it with the PR vsix installer if you like.

PR #1131 (2024-05-21): Bugfix/debugging docs
## Description

Update documentation regarding debugging using other extensions and now using Eclipse CDT GDB

Fixes #918
Fixes #1164 
Fixes #1138 
Fixes #1168 
Fixes #1177

## Type of change
- This change requires a documentation update
- New feature

## Steps to test this pull request

1. Create a new ESP-IDF Project
2. Build flash the project
3. Use the CDT debug configuration to start debugging.

```json
 {
   "configurations": [
     {
       "type": "gdbtarget",
       "request": "attach",
       "name": "Eclipse CDT Remote",
       "program": "${workspaceFolder}/bu [... cut]
closed 2024-04-13 by github-actions[bot]