espressif/vscode-esp-idf-extension #1141: SDK Config won't work as documented on FREERTOS Kernel (configUSE_TRACE_FACILITY  configUSE_STATS_FORMATTING_FUNCTIONS  Enable display of xCoreID in vTaskList  configGENERATE_RUN_TIME_STATS)

CescG, 2024-02-18:
### OS

MacOS

### Operating System version

Sonoma 14.3.1

### Visual Studio Code version

1.86

### ESP-IDF version

5.1.1

### Python version

3.11.5

### Doctor command output

 /Library/Frameworks/Python.framework/Versions/3.11/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin 

### Description

SDK Configuration parameters FREERTOS/Kernel cannot be set properly.

Cannot set configUSE_TRACE_FACILITY nor Enable display of xCoreID in vTaskList nor Enable display of xCoreID in vTaskList
without setting configUSE_STATS_FORMATTING_FUNCTIONS too.

In fact, it CAN bet seated using the SDK configuration editor but once saved, if retrieved the SDK editor again, configUSE_STATS_FORMATTING_FUNCTIONS is always set.  According my understanding of the documentation it should not be set.

thanks for your help

### Debug Message

```plain
..
```

### Other Steps to Reproduce

I can do it agin if you needed

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-02-20:
I've tested with ESP-IDF version 5.2.0 on latest master and it does work to enable just configUSE_TRACE_FACILITY.

This doesn't seem to be an extension issue. I will suggest to review the Output -> ESP-IDF and see what happens when you select configUSE_TRACE_FACILITY, the confserver output should provide more detail to as why is it failing for your version.

brianignacio5, 2024-03-06:
Closing this issue for lack of response. Please comment here if you still have issues using feature.

closed 2024-03-06 by brianignacio5