espressif/vscode-esp-idf-extension #1149: Can't configure extension without pip installed

timoxd7, 2024-02-28:
### OS

Linux

### Operating System version

Ubuntu 22.04

### ESP-IDF version

5.1.2

### Python version

3.8.10

### Doctor command output

-

### Description

Can't configure the extension if pip is not installed globally.
This is the case if using the Docker Image of ESP IDF to create a Development Container (e.g. Github Codespaces). The extension will search for python and pip and then start installation. Inside the stock Docker Image of ESP IDF, there is no pip globally installed, only in venv.

However, if the extension is preconfigured in settings, the pip check will not happen and the extension will work as expected.

Also see espressif/esp-idf#13266, workaround is to install pip in docker container globally, which however should not be the recommended way.

### Debug Message

```plain
Can't find pip
```

### Other Steps to Reproduce

1. Create Development Container
2. Start VS Code inside of it
3. Start Configuration of Extension

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-02-28:
The idea is that the extension doesn't need to be configured when you are using a docker container. The settings in .devcontainer.json should provide you with already configured example.

Let's try this, in your current project in your host computer use the `ESP-IDF: Add Docker Container Configuration` command. This will add a `.devcontainer` folder in your project. Now try to use `Dev Containers: Reopen in Container` command to reopen folder in based on these template container configuration.

brianignacio5, 2024-03-11:
Closing this issue for lack of response. 

Please comment here if you need more assistance.

closed 2024-03-11 by brianignacio5