espressif/vscode-esp-idf-extension #1150: compiler broken during compile esp32 example in Windows

JunyounsKim, 2024-02-28:
### OS

Windows

### Operating System version

Windows 10

### ESP-IDF version

5.2

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\Intel\iCLS Client\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Samsung\SamsungLink\AllShare Framework DMS\bin\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Program Files\Amazon\AWSCLIV2\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\PuTTY\;C:<HOMEPATH>\AppData\Local\Programs\Python\Launcher\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Roaming\npm;C:\Program Files\heroku\bin;C:<HOMEPATH>\AppData\Local\Programs\Hyper\resources\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.86.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.0 
Workspace folder c:<HOMEPATH>\test1 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.11.1;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM3
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.11.1: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
[... 169 lines cut ...]
    C:/Users/김주년/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: cannot find -lgcc: No such file or directory
    C:/Users/김주년/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: cannot find -lc: No such file or directory
    C:/Users/김주년/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: cannot find -lnosys: No such file or directory
    C:/Users/김주년/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: cannot find -lc: No such file or directory
    C:/Users/김주년/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: cannot find -lgcc: No such file or directory
    C:/Users/김주년/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: cannot find C:/Users/김주년/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/esp32/crtend.o: No such file or directory
    C:/Users/김주년/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: cannot find C:/Users/김주년/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/esp32/crtn.o: No such file or directory
    collect2.exe: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  C:/Users/김주년/esp/esp-idf/tools/cmake/project.cmake:506 (__project)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/김주년/Documents/idf_example/blink/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/김주년/Documents/idf_example/blink/build/CMakeFiles/CMakeError.log".

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\Users\김주년\Documents\idf_example\blink\build -S c:\Users\김주년\Documents\idf_example\blink" terminated with exit code: 1.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-02-28:
Maybe try to delete `C:/Users/김주년/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928` and reinstall the ESP-IDF Tools ?

JunyounsKim, 2024-02-28:
After deleting `C:/Users/김주년/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928` folder and reinstalling, it still shows same broken error. Is there any problem in that compiler version?

srk517, 2024-03-03:
Hello I am facing same issue currently but I am using Ubuntu. Can anyone help?
.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: cannot find -lphy: No such file or directory

brianignacio5, 2024-03-21:
Closing this issue for lack of response. Comment here if you still have any issues.

closed 2024-03-21 by brianignacio5