espressif/vscode-esp-idf-extension #1154: Question: is there a way to disable the ESP-IDF plugin Welcome screen in VSC?

mbratch, 2024-03-02:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.7.0

### ESP-IDF version

5.1.2

### Python version

3.11.2

### Doctor command output

NA

### Description

Every time I open an existing ESP-IDF project in VSC, I will start working on a source file and then the annoying ESP-IDF plugin Welcome tab opens, interrupting what I was doing. It takes awhile for the tab to appear after opening a project so it adds considerable delay to working after opening a project.

Is there a way to turn off this welcome tab from opening automatically?

### Debug Message

```plain
NA
```

### Other Steps to Reproduce

Just open an existing ESP-IDF project and wait.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mbratch, 2024-03-04:
Sorry... I finally found some solutions after further searching.

closed 2024-03-04 by mbratch

brianignacio5, 2024-03-06:
You could also set the `idf.showOnboardingOnInit: false` in your .vscode/settings.json or User settings

If you find the extension useful. Please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)

RobinKayGeoMoby, 2024-04-24:
> Sorry... I'm totally blind and I was being daft. There is a checkbox in the upper left corner. Sheesh. On a related note, I did try setting startupEditor option to "none" but that didn't block the welcome tab.

I have the same problem and not seeing any checkbox.

brianignacio5, 2024-04-24:
[image]

You could also set the `idf.showOnboardingOnInit: false` in your .vscode/settings.json or User settings

RobinKayGeoMoby, 2024-04-25:
Thanks Brian,

I appreciate the follow-up.

Regards,
Robin

riscy00, 2024-04-30:
This worked