espressif/vscode-esp-idf-extension #1155: Unable to read the Core Dump file

julianiaccopucci, 2024-03-05:
### OS

MacOS

### Operating System version

Somona 14.3.1 

### Visual Studio Code version

1.87.0

### ESP-IDF version

5.1.2

### Python version

3.12.1

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 23.3.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /opt/homebrew/opt/openjdk@17/bin:<HOMEPATH>/.rbenv/shims:/Library/Frameworks/Python.framework/Versions/3.12/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/Apple/usr/bin:<HOMEPATH>/.rvm/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.87.0 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.7.0 
Workspace folder <HOMEPATH>/br-firmware 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.1_py3.12_env/bin/python
Serial port (idf.port) /dev/cu.usbserial-110
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) /usr/bin/git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
[... 138 lines cut ...]

It says that the GDB executable is not found.

I've followed all the documentation available but I'm unable to get around this. I've even added the xtensa binaries to my path and I can call them from my terminal window from any path. 

Other info:
- The command "Launch IDF Monitor for CoreDump / GDB-Stub Mode" works fine.

### Debug Message

```plain
julian@Julians-MBP br-firmware % /Users/julian/.espressif/python_env/idf5.1_py3.12_env/bin/python '/Users/julian/esp/esp-idf/tools/idf.py' coredump-info -b 115200
Setting IDF_PATH environment variable: /Users/julian/esp/esp-idf
Executing action: coredump-info
Serial port /dev/cu.usbserial-110
Connecting....
Detecting chip type... ESP32-S3
===============================================================
==================== ESP32 CORE DUMP START ====================
GDB executable not found. Please install GDB or set up ESP-IDF to complete the action. Read more: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

julianiaccopucci, 2024-03-05:
esp_idf_vsc_ext.log

{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:27)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:27)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /dist/extension.js:2:1396887\n\tat ChildProcess.exithandler (node:child_process:412:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat maybeClose (node:internal/child_process:1091:16)\n\tat Process.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2024-03-05T08:16:01.863Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:27)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:27)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /dist/extension.js:2:1396887\n\tat ChildProcess.exithandler (node:child_process:412:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat maybeClose (node:internal/child_process:1091:16)\n\tat Process.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2024-03-05T08:45:36.609Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:27)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:27)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /dist/extension.js:2:1396887\n\tat ChildProcess.exithandler (node:child_process:412:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat maybeClose (node:internal/child_process:1091:16)\n\tat Socket.<anonymous> (node:internal/child_process:449:11)\n\tat Socket.emit (node:events:514:28)\n\tat Pipe.<anonymous> (node:net:323:12)","level":"error","timestamp":"2024-03-05T08:45:36.827Z"}

brianignacio5, 2024-03-05:
Are you running the command `idf.py coredump-info -b 115200` from `ESP-IDF: Open ESP-IDF Terminal ` command ? or just a regular terminal ? The gdb is loaded for the first case but not the second.

julianiaccopucci, 2024-03-05:
I was clicking on the 'Terminal' tab of VSCode, rather than `ESP-IDF: Open ESP-IDF Terminal`. Now it works fine, thanks.

brianignacio5, 2024-03-06:
Glad it works !

If you find the extension useful. Please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)