espressif/vscode-esp-idf-extension #1164: JTAG debugging failure

mukymuk, 2024-03-19:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.87.2

### ESP-IDF version

5.1.2

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\MATLAB\R2023b\bin;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Windows\System32\OpenSSH;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files (x86)\Nmap 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.87.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.0 
Workspace folder c:<HOMEPATH>\project\bh\asv\fw\asv\hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Espressif\frameworks\esp-idf-v5.1.2
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) board/esp32c2-ftdi.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif
Git Path (idf.gitPath) C:\Espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:\Espressif\tools\cmake\3.24.0\bin: true
Access to C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to C:\Espressif\tools\ninja\1.10.2: true
Access to C:\Espressif\tools\idf-exe\1.0.3: true
Access to C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
[... 717 lines cut ...]
Traceback (most recent call last):
  File "threading.py", line 1038, in _bootstrap_inner
  File "threading.py", line 975, in run

[Debug Adapter]
  File "threading.py", line 975, in run
  File "subprocess.py", line 1552, in _readerthread
OSError: [Errno 22] Invalid argument
  File "subprocess.py", line 1552, in _readerthread
OSError: [Errno 22] Invalid argument

[Debug Adapter]
0x3fcad8f4
0x3fcad8f0
0x3fcad8f4
0x3fcad8f0
DEBUG_ADAPTER_STOPPED
2024-03-18 21:43:46,749 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED

[Stopped] : ESP-IDF Debug Adapter
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-03-22:
We are working on migrating to a new debug adapter that is more stable that our current python one in this PR:

https://github.com/espressif/vscode-esp-idf-extension/pull/1131

You are welcome to test it with the PR vsix installer if you like.

mukymuk, 2024-03-22:
I tried the vsix and got an error during install the first attempt.  Second attempt seemed to install, but nothing new showed up in the extensions list so I don't know if it's installed or not.  Various debugging errors remain.  JTAG Debugging seems so utterly broken in vscode and in eclipse that it's amazing to me that you can even get it to pass basic functional tests at all.

brianignacio5, 2024-03-22:
Did you tried the new launch.json configuration ? Maybe you are still using the old debug adapter if you don't update the launch.json content. Something like this. (was also updated in docs in the PR)

```JSON
{
   "configurations": [
     {
       "type": "gdbtarget",
       "request": "attach",
       "name": "Eclipse CDT Remote",
       "program": "${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
       "initCommands": [
         "set remote hardware-watchpoint-limit 2",
         "mon reset halt",
         "maintenance flush register-cache",
         "thb app_main",
       ],
       "gdb": "${command:espIdf.getXtensaGdb}",
       "target": {
         "port": "3333"
       },
     }
   ]
 }
```

mukymuk, 2024-03-22:
That was the problem.  Thanks!  There's still some errors here and there, but I have basic debug functionality now.

PR #1131 (2024-05-21): Bugfix/debugging docs
## Description

Update documentation regarding debugging using other extensions and now using Eclipse CDT GDB

Fixes #918
Fixes #1164 
Fixes #1138 
Fixes #1168 
Fixes #1177

## Type of change
- This change requires a documentation update
- New feature

## Steps to test this pull request

1. Create a new ESP-IDF Project
2. Build flash the project
3. Use the CDT debug configuration to start debugging.

```json
 {
   "configurations": [
     {
       "type": "gdbtarget",
       "request": "attach",
       "name": "Eclipse CDT Remote",
       "program": "${workspaceFolder}/bu [... cut]
closed 2024-04-13 by github-actions[bot]