espressif/vscode-esp-idf-extension #1166: can't change port on macos

mianos, 2024-03-21:
### OS

MacOS

### Operating System version

Sonoma 14.4

### Visual Studio Code version

1.87.2

### ESP-IDF version

v5.2.1

### Python version

3.11.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin x64 23.4.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.local/adb:/usr/local/opt/llvm/bin:/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/bin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/usr/local/go/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.87.2 
Visual Studio Code language en 
Visual Studio Code shell /bin/bash 
ESP-IDF Extension version 1.7.0 
Workspace folder <HOMEPATH>/wa/ETM/gptimer_capture_hc_sr04 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/v5.2.1/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.2_py3.11_env/bin/python
Serial port (idf.port) /dev/tty.wchusbserial57350193411
OpenOCD Configs (idf.openOcdConfigs) board/esp32c6-ftdi.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
[... 119 lines cut ...]
### Description

When selecting serial port it does not change.

### Debug Message

```plain
{"user":true,"message":"Something went wrong while getting the serial port list","stack":"TypeError: Cannot read properties of undefined (reading 'uri')\n\tat h (/dist/extension.js:2:1346779)\n\tat async kt.updatePortListStatus (/dist/extension.js:2:1427032)\n\tat async kt.displayList (/dist/extension.js:2:1426820)","level":"error","timestamp":"2024-03-21T00:51:25.385Z"}
```

### Other Steps to Reproduce

Waveshare ESP32C6 with combined jtag and serial loader ports.
 
In the end I just ran openocd manually to get the bootloader installed and then it seems the random usb to serial adapter worked.

     ./bin/openocd -f share/openocd/scripts/board/esp32c6-builtin.cfg

I have looked at several other instances of this issue and they say something about different user settings overrides but they don't actually say the clue how to fix it. I tried setting the local port, and deleting it but then somehow it got hard coded to /dev/ttyUSB0 and I could never get it back until I again hard coded it into the local project settings.

I am not even sure if the openocd start command line set up by the extension is functional.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-03-21:
After you select the serial port, there should be a dropdown to choose the workspace folder to save the setting. Do you see this dropdown ?

mianos, 2024-03-21:
wow, ok. I don't know why that was not obvious. 

Then when I don't select that and click away it ignores my selection with this warning.

Thanks. At least this answers makes is much clearer than the other two.

That's quite a non intuitive pattern. I guess it's an MSVC UI framework limitation.

(Also guess it's like the whole IDF install instructions that don't mention everything is driven from the buttons along the bottom row. Or the fact that you actually have to select the IDF version in the initial install page).

Thanks again.

closed 2024-03-21 by mianos

brianignacio5, 2024-03-21:
Will try to improve this so it chooses the already selected workspace folder.

If there something else not clear from documentation or UI, please let us know so we can improve.

brianignacio5, 2024-04-08:
If you find the extension useful, please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension) we haven't received any GitHub issue regarding setup problem as mentioned.