espressif/vscode-esp-idf-extension #1170: CMakeList.txt being overwritten during normal compile

mianos, 2024-03-22:
### OS

MacOS

### Operating System version

Sonoma 14.4

### Visual Studio Code version

1.87.2

### ESP-IDF version

v5.2.1

### Python version

3.11.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin x64 23.4.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.local/adb:/usr/local/opt/llvm/bin:/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/bin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/usr/local/go/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.87.2 
Visual Studio Code language en 
Visual Studio Code shell /bin/bash 
ESP-IDF Extension version 1.7.0 
Workspace folder <HOMEPATH>/wa/hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/v5.2.1/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.2_py3.11_env/bin/python
Serial port (idf.port) /dev/cu.wchusbserial57350193411
OpenOCD Configs (idf.openOcdConfigs) board/esp32c6-ftdi.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
[... 134 lines cut ...]
```
Then after a bunch of compiles I was getting main_task+0x76): undefined reference to `app_main'
I looked at it:
```
idf_component_register(SRCS
                    INCLUDE_DIRS "")
```
If I put the file name back in, it now obviously works.

This has happened 3 times now. Is there a button near build in the IDF explorer panel that does this that I may be hitting?

### Debug Message

```plain
none
```

### Other Steps to Reproduce

Just compile, compile and upload.
I'll add anything else I can here if I see it again.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-03-25:
There is a setting called `idf.enableUpdateSrcsToCMakeListsFile` which is true by default which updates the CMakeLists.txt automatically based when you add or delete source files in your project.

You can disable it by setting it to false in your `.vscode/settings.json`.

Could you describe steps to reproduce what did you do with sources ? We haven't received any issue with this feature.

mianos, 2024-03-25:
I didn't do anything but edit, compile debug. But I am using vi outside of the editor. Maybe as vi was swapping the filename on a normal backup and write, the file was gone from the directory for a tiny moment and the caused it to be removed.
I'll close this with this plausible explanation and adjust the settings to prevent this from happening. I use cmake at work so I think manually editing it would be better for me anyway.

mianos, 2024-03-25:
ps, thanks.

closed 2024-03-25 by mianos