espressif/vscode-esp-idf-extension #1175: project fails to build only with vscode-esp-idf-extension under MacOS arm64 - linux, windows, macos idf.py works fine

mhaberler, 2024-04-02:
### OS

MacOS

### Operating System version

14.4.1

### Visual Studio Code version

1.87.2

### ESP-IDF version

5.2.1 release

### Python version

3.12.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 23.4.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.nvm/versions/node/v18.16.1/bin:<HOMEPATH>/.cargo/bin:<HOMEPATH>/go/bin:/usr/local/go/bin:/opt/homebrew/bin:/usr/local/bin:/bin:/opt/homebrew/sbin:/usr/local/sbin:/usr/bin:/usr/sbin:/sbin:/opt/X11/bin:/Applications/Wireshark.app/Contents/MacOS:/Applications/Visual Studio Code.app/Contents/Resources/app/bin:<HOMEPATH>/.local/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.87.2 
Visual Studio Code language en 
Visual Studio Code shell /opt/homebrew/bin/bash 
ESP-IDF Extension version 1.7.1 
Workspace folder <HOMEPATH>/Ballon/src/BalloonWare/idf/esp-idf-wss-https-server 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/v5.2.1/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.11.1:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.2_py3.12_env/bin/python
Serial port (idf.port) /dev/cu.usbmodem11101
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) /opt/homebrew/bin/git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin: false
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/ninja/1.11.1: false
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
[... 144 lines cut ...]
CMake Error at /Users/mah/esp/v5.2.1/esp-idf/tools/cmake/project.cmake:506 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32s3-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:66 (project)

-- Warning: Did not find file Compiler/-ASM
```

### Other Steps to Reproduce

clone  https://github.com/mhaberler/esp-idf-wss-https-server/
open in VSCode MacOS Mx (arm)
build witth vscode-esp-idf-extension 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-04-03:
Seems to me you have a shell configuration issue since your the build task can't find the executables from the extension appended PATH. I suspect it might be related to your configured shell `/opt/homebrew/bin/bash`.

I would suggest using a macOS default or make sure that the PATH configuration works.

mhaberler, 2024-04-03:
Assume this were the case: 

then both a manual build from bash, as well as a manual build from "ESP-IDF: Open ESP-IDF Terminal" of this example project should fail as well, correct?

well it doesnt - those builds succeed and flash perfectly - as described above.

Can I kindly ask you to try to reproduce to build the example repo on an MacOs Mx arm64 platform with the vscode-esp-idf-extension - which is the only platform where this fails?

thanks in advance

Michael

mhaberler, 2024-04-05:
I created a test user in MacOS using default zsh, and installed esp-idf, vscode, and this extension there
this actually built the test repo linked above without errors

switching my main account to zsh now, and reinstalling homebrew - let's see

brianignacio5, 2024-04-08:
Glad it works !

If you find the extension useful. Please leave us a review in the [Visual Studio Code Marketplace](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension)

closed 2024-04-08 by brianignacio5