espressif/vscode-esp-idf-extension #1180: ESP-IDF extension for VSCod "zip file error" on install

pegeco, 2024-04-15:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.88.1

### ESP-IDF version

5.0.6

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.88.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.0 
Workspace folder c:<HOMEPATH>\Documents\GitHub\eeva 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/frameworks/esp-idf-v5.0.4/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Espressif\tools\esp-rom-elfs\20220823
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Espressif\tools\esp-rom-elfs\20220823/
Virtual env Python Path (idf.pythonBinPath) C:/Espressif/python_env/idf5.0_py3.11_env/Scripts/python.exe
Serial port (idf.port) COM48
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif
Git Path (idf.gitPath) C:/Espressif/tools/idf-git/2.39.2/cmd/git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:\Espressif\tools\cmake\3.24.0\bin: true
Access to C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin: true
Access to C:\Espressif\tools\ninja\1.10.2: true
Access to C:\Espressif\tools\idf-exe\1.0.3: true
Access to C:\Espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
[... 127 lines cut ...]
Error is: 
_Zip file error
Source: ESP-IDF_

### Debug Message

```plain
idf-git-2.39.2-win64.zip progress: 91.05% (25973.29 / 28526.53) KB
idf-git-2.39.2-win64.zip progress: 91.11% (25989.29 / 28526.53) KB
idf-git-2.39.2-win64.zip progress: 91.16% (26005.29 / 28526.53) KB
Error: aborted
	at connResetException (node:internal/errors:720:14)
	at TLSSocket.socketCloseListener (node:_http_client:461:19)
	at TLSSocket.emit (node:events:529:35)
	at node:net:350:12
	at TCP.done (node:_tls_wrap:614:7)
Failed download. Retrying...
Error: Error: HTTP/HTTPS Response error
Installing c:\Espressif\tools\idf-git\2.39.2 ...
Zip file error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-04-17:
Seems like you have a network issue ( or did you cancelled the setup wizard during download?) I will suggest to review the network connection and please try again.

Also review if your system or VSCode is configured to use a proxy server and make sure it is working.

pegeco, 2024-04-17:
> Seems like you have a network issue ( or did you cancelled the setup wizard during download?) I will suggest to review the network connection and please try again.
> 
> Also review if your system or VSCode is configured to use a proxy server and make sure it is working.

Apart from taking this laptop home and using my home network (which previously worked and allowed me to download fully) - is there a workaround for this? Maybe I can download/unzip that file and others manually, for example?

closed 2024-05-09 by github-actions[bot]

jargonised, 2025-10-02:
Please, don't give advice in the style of "is your computer turned on" because if someone came to this one error, it's obvious they had Internet connection and they won't post here without trying few times.

I have the same error on Ubuntu. I have python 3.13 but esp-idf v.5.5. keeps informing me I have lower version than the required 3.9.

So I've tried v.5.4.2 - and then I have the mentioned issue (zip file issue).