espressif/vscode-esp-idf-extension #1181: expressif fmt download and reported bug related to cstddef in fmt/code.h

riscy00, 2024-04-15:
### OS

Windows

### Operating System version

Window 11

### Visual Studio Code version

1.35.1

### ESP-IDF version

4.4.3

### Python version

3.8.7

### Doctor command output

ESP32H2 DevKit Board with two USB-C port (Aliexpress, working USB/JTAG) with blink example project via ESP-IDF 5.2.1.  
Clean and Build no problem.
I then use component add and installed expressif fmt library
I added #include "fmt/core.h" as per guide. 
I then clean and build and it come up with this error.  

### Extension

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/14983388/esp_idf_vsc_ext.log)

### Description

Unable to complete compiler task due to issue with the library related to fmt, which could not find #include <cstddef>, I checked and it did exist in the 13.2.0 but not found, not sure why? 

C:\Users\riscy\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\riscv32-esp-elf\include
C:\Users\riscy\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\riscv32-esp-elf\include\c++\13.2.0

Unable to use fmt, can you advise?

### Debug Message

```plain
In file included from E:/016_Work_ESP32/010_Example/BlinkH2_ESPIDF/blink/main/blink_example_main.c:16:
E:/016_Work_ESP32/010_Example/BlinkH2_ESPIDF/blink/managed_components/espressif__fmt/fmt/include/fmt/core.h:11:10: fatal error: cstddef: No such file or directory
   11 | #include <cstddef>  // std::byte
      |          ^~~~~~~~~
compilation terminated.
```

### Other Steps to Reproduce

Went thro google search, gemini, bing copilot and could not figure out the solution. 
Have change between #include <fmt/core.h>   and #include "fmt/core.h", no improvement. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

riscy00, 2024-04-15:
It seems that the project has to be C++, not C, as in the Blink demo code. I would have to make a new project that is based on C++ so it works correctly with the fmt component. Do you have Blink demo code in C++ or similar demo variants? If possible do you have guideline dealing with mixed C and C++ and how to solve problem. I have lot of experience in embedded C but not much in C++. Thanks.

brianignacio5, 2024-04-17:
How is your C/C++ configuration file ? Are you using the Microsoft C/C++ extension or Clang or what for language syntax ? Are you using compile_commands.json ? The doctor command could provide such help

riscy00, 2024-04-17:
I'm using ESP-IDF extension (5.2.1) installed in Window 11 within VSC, I believe it using gcc but I check, I have studied into CMAKE and NINJA (with MiniGW for gcc/g++) for past few days and getting familiar with it. I still not sure what is the right way to specify or include G++ along with GCC library in CMAKE or JSON setting, so that error goes away.

brianignacio5, 2024-04-18:
Please take a look at this documentation: https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md

The extension need to set `compile_commands.json` to resolve these definitions.

riscy00, 2024-04-18:
[](url)[fmt_bug_issue.pdf](https://github.com/espressif/vscode-esp-idf-extension/files/15026055/fmt_bug_issue.pdf)
Please see attached, I have followed the instruction and modified it, the red wiggle under the #include <fmt/core.h> went away but I still get same error: cstddef
I have attached details in PDF and compiler logs if this any help
[terminalLogB.txt](https://github.com/espressif/vscode-esp-idf-extension/files/15026137/terminalLogB.txt)

riscy00, 2024-04-26:
It turned out the issue is C/C++ lang linkage. I have it sorted out. The FMT must be treated as CPP not as C.  The CMake file generated by the ESP-IDF looks good after a few days of reading and learning.

riscy00, 2024-04-26:
Closed case for now.

closed 2024-04-26 by riscy00