espressif/vscode-esp-idf-extension #1185: Failed setup in VSCode due to missing pip - Arch Linux

QuantumQuacken, 2024-04-29:
### OS

Linux

### Operating System version

ArcoLinux x64 6.8.7-arch1-2 

### Visual Studio Code version

1.88.1

### ESP-IDF version

5.4.0

### Python version

3.12.3

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 6.8.7-arch1-2 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /sbin:/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/sbin:/var/lib/flatpak/exports/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/var/lib/snapd/snap/bin 
System environment variable PYTHON 
 /usr/bin/python 
Visual Studio Code version 1.88.1 
Visual Studio Code language en 
Visual Studio Code shell /sbin/bash 
ESP-IDF Extension version 1.7.1 
Workspace folder <HOMEPATH>/Documents/Git/arduino 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars)
Virtual env Python Path (idf.pythonBinPath) python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) /usr/bin/git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.44.0
ESP-IDF version 5.4.0
Python version 3.12.3
Python's pip version Not found
-------------------------------------------------- Project configuration settings ----------------------------------------------------------
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF Doctor Command
--------------------------------------------------------------------------------------------------------------------------------------------

### Description

Hello,
I'm following the installation [tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md) in this repo, but I'm stuck at the setup phase within VSCode. I've already installed ESP-IDF in my HOME dir, so I'm choosing the "Use existing setup" option, but, after selecting all the right options and pressing "Install", I receive the message ""/usr/bin/python -m pip" is not valid. (ERROR_INVALID_PIP)". The fact that pip is missing is normal, because python packages are externally managed. Is there a way to bypass this issue in the setup process or use a more appropriate alternative based on my environment?

### Debug Message

```plain
"/usr/bin/python -m pip" is not valid. (ERROR_INVALID_PIP)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

OrhanYigitDurmaz, 2024-05-01:
i have installed pip using pacman in some way, and installed the esp-idf. however when i got to build an example, ninja just said no work to do. looking at the logs, i can see the extension looking for `idf5.2_py3.11_env` but i only had `idf5.2_py3.12_env`.

OrhanYigitDurmaz, 2024-05-01:
running `/home/orhan/.espressif/python_env/idf5.2_py3.12_env/bin/python3.12 /home/orhan/esp/v5.2.1/esp-idf/tools/idf.py --list-targets
` works

OrhanYigitDurmaz, 2024-05-01:
okay uhmmmm so i forgot that vscode keeps a settings.json file and the python bin dir setting was there too i forgot to change that... now that i have done that nowi can list targets from vscode, however i cant set them up. 

```[Error: non zero exit code 2

Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory /home/orhan/Documents/GitHub/ESP32Code/firmware/TwistyFirmware-esp-idf/twisty_standalone/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/orhan/.espressif/python_env/idf5.2_py3.12_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/orhan/Documents/GitHub/ESP32Code/firmware/TwistyFirmware-esp-idf/twisty_standalone"...
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /sbin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /sbin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Existing sdkconfig '/home/orhan/Documents/GitHub/ESP32Code/firmware/TwistyFirmware-esp-idf/twisty_standalone/sdkconfig' renamed to '/home/orhan/Documents/GitHub/ESP32Code/firmware/TwistyFirmware-esp-idf/twisty_standalone/sdkconfig.old'.
-- Found Git: /sbin/git (found version "2.45.0")
-- Configuring done (1.4s)
-- Generating done (0.0s)
CMake Warning:
  Manually-specified variables were not used by the project:

    CCACHE_ENABLE
    ESP_PLATFORM

-- Build files have been written to: /home/orhan/Documents/GitHub/ESP32Code/firmware/TwistyFirmware-esp-idf/twisty_standalone/build
Cannot load /home/orhan/Documents/GitHub/ESP32Code/firmware/TwistyFirmware-esp-idf/twisty_standalone/build/project_description.json: [Errno 2] No such file or directory: '/home/orhan/Documents/GitHub/ESP32Code/firmware/TwistyFirmware-esp-idf/twisty_standalone/build/project_description.json'

	at ChildProcess.<anonymous> (/dist/extension.js:2:1394091)
	at ChildProcess.emit (node:events:517:28)
	at ChildProcess._handle.onexit (node:internal/child_process:292:12)]
```
this is the output i get.

OrhanYigitDurmaz, 2024-05-01:
uhhh fixed it by deleting settings.json file inside .vscode folder

brianignacio5, 2024-05-13:
Well, as described in the documentation, pip is part of ESP-IDF prerequisite before even installing the extension (since it is handled by the OS not the extension)/

closed 2024-05-13 by brianignacio5