espressif/vscode-esp-idf-extension #1186: idf.py command from inside Visual Studio Code only returns 'Please use idf.py only in an ESP-IDF shell environment'. (IDFGH-12776)

saillingaway, 2024-05-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### IDF version.

v5.1.1-dirty

### Operating System used.

Mac OS Ventura 13.4

### How did you build your project?

Command line with idf.py build

### What is the expected behavior?

`idf.py build` or other subcommands like `flash` or `monitor` should run.

### What is the actual behavior?

"Please use idf.py only in an ESP-IDF shell environment" is outputted to the terminal.
[image]

### Technical Information.
Machine: M2 Mac
Visual Studio Code Version: 1.85.2

### More Information.

I recently encountered this issue suddenly that I didn't have at all before.

For my project that uses ESP-IDF I usually use the `get_idf` alias and then the `idf.py` commands like `build flash monitor` from inside vscode's terminal. I took a break from the project for a month or so and I opened it back up again just the other week and now all the `idf.py` commands only result with the message **Please use idf.py only in an ESP-IDF shell environment**, even though running `get_idf` shows output that makes it seems like everything in the `install.sh` file is working just fine.

If I open a standalone terminal, running my `get_idf`alias followed by any `idf.py` commands will work as expected (if I'm in a project directory). 

I understand that there must be _some_ difference between the standalone terminal and the one inside vs code, however I don't know how to figure out what that difference is, or how to adjust my vs code terminal to become an "ESP-IDF shell environment".nIf anyone can help me understand this issue better and/or fix it I'd appreciate it, thanks in advance.

brianignacio5, 2024-05-21:
I'm not sure why this happens. From esp-idf itself it is generated when idf.py try to import idf_py_actions and other python packages. it might be a indicator that your esp-idf or setup might be missing something but without env information is hard to know.

saillingaway, 2024-05-21:
@brianignacio5 Okay, thanks for the ref. If you don't mind leaving this open for a bit, I'll come back and share my findings once I have time to troubleshoot this more in depth.

brianignacio5, 2024-05-22:
You could also try using this extension and use the `Open ESP-IDF Terminal` command to launch a terminal. Should be configured if extension is configured.

closed 2024-06-13 by github-actions[bot]