espressif/vscode-esp-idf-extension #1192: Split compile errors and warnings in output[Feature Request]:

Superberti, 2024-05-17:
**Is your feature request related to a problem? Please describe.**
Hi, when a project gets larger it's quite common that you have a bunch of warning messages and errors in the compile output. It's OK to show the warnings but often they will be addressed later. Sometimes it is a bit messy to find the errors in the large output window between warnings, compiler messages and errors.

**Describe the solution you'd like**
A dedicated warnings tab and an dedicated errors tab would be much more convenient.

**Describe alternatives you've considered**
I searched around in the net but could not find something in this direction. The "Problems" tab seems to be IntelliSense based and does not show the "real" errors of all project files.

brianignacio5, 2024-05-21:
What you are looking is a vscode feature itself as described in here:

https://code.visualstudio.com/updates/v1_23#_problems-view-filtering

Superberti, 2024-05-21:
Hi,

as I wrote before, the "Problems" tab does NOT show all compile errors and warnings, so unfortunately this is not a solution.

brianignacio5, 2024-05-21:
Hi @Superberti the title is to `Split compile errors and warnings`  and the described solution you would like so as I understand this checkmark should achieve this by not showing Warnings or Errors, etc.

With regards to not showing all compile errors and warnings, could you provide an example or steps to reproduce this behavior. Which errors or warning are not showing ?

Superberti, 2024-05-21:
Hi again,

after some testing I can't reproduce the described behavior of the "Problems" tab. Maybe I got a wrong impression of the tab when sources with warnings has been already compiled and the warnings will not appear again in this case (which is quite normal).  
Building a clean solution will fill the problems tab as expected, so let's close this feature request as it's already here.

vicmaweu, 2024-09-12:
I believe what  [Superberti](https://github.com/Superberti) was describing was a way which to like tab the errors and warnings separately just like how Microsoft Visual Studio IDE's do. Like below: 

[image]

brianignacio5, 2024-09-13:
> I believe what [Superberti](https://github.com/Superberti) was describing was a way which to like tab the errors and warnings separately just like how Microsoft Visual Studio IDE's do. Like below:
> 
> ![Capture](https://private-user-images.githubusercontent.com/119595100/366741890-804e13f7-018f-4dff-9bc3-3cd538b7939a.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYyMDkxODYsIm5iZiI6MTcyNjIwODg4NiwicGF0aCI6Ii8xMTk1OTUxMDAvMzY2NzQxODkwLTgwNGUxM2Y3LTAxOGYtNGRmZi05YmMzLTNjZDUzOGI3OTM5YS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxM1QwNjI4MDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zNTAxMjM2MmY2Njc4YzdjYjJmMjAwZDZlZDhlODM5MjBhMThjZTNkNmM3MjNjNDMyMDc2ZTIxMTI4ZWMwZjEyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.QlNroguB5GWpvxLLM3Iy1RNStcrvkBToyp57atD5TC8)

So basically this : https://github.com/espressif/vscode-esp-idf-extension/issues/1192#issuecomment-2122169131 ?

Superberti, 2024-09-13:
Yes, this would be nice. 
And I recognized, that some types of errors in the "terminal" tab are not shown in the "problems" tab. But I can't remember exactly if this was on linker errors or problems in the build system. Next time I keep an eye on this!