espressif/vscode-esp-idf-extension #1193: "Show all ESP-IDF tags" not working

gonzabrusco, 2024-05-18:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.89.1 (latest)

### ESP-IDF version

4.4.4

### Python version

Latest

### Doctor command output

No applicable

### Description

When trying to configure the IDF extension, when I mark the option "Show all ESP-IDF tags" it does not update the list of available IDF versions to install. I want to install version 4.4.4 but it does not show in the list.

### Debug Message

```plain
No applicable
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

gonzabrusco, 2024-05-18:
[image]

brianignacio5, 2024-05-20:
I'm not sure why but GitHub api https://api.github.com/repos/espressif/esp-idf/tags just show these tags.

gonzabrusco, 2024-05-20:
Strange! Is there a way to force the extension to use 4.4.4? Or at least could you tell me how to do it manually?

gonzabrusco, 2024-05-21:
I've found the problem @brianignacio5 

Github is limiting the amount of entries per request. You have to specify if you need more! 

For example:
https://api.github.com/repos/espressif/esp-idf/tags?per_page=5000

Doc:
https://docs.github.com/es/rest/using-the-rest-api/using-pagination-in-the-rest-api?apiVersion=2022-11-28

PR #1196 (2024-05-22): Fix Github tag url to receive 100 tags (VSC-1357)
## Description

Adds more tags to the github request.

Fixes #1193

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Checklist
- [ ] PR Self Reviewed
- [ ] Applied Code formatting
- [ ] Added Documentation
- [ ] Added Unit Test
- [ ] Verified on all platforms - Windows,Linux and macOS
closed 2024-05-22 by brianignacio5