espressif/vscode-esp-idf-extension #1194: Something went wrong while trying to build the project, serial port don't work and after full system restart

DawidGurdzinski2, 2024-05-19:
### OS

Windows

### Operating System version

Windows11

### Visual Studio Code version

1.89.1

### ESP-IDF version

5.1.1

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\CMake\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312\;C:<HOMEPATH>\AppData\Local\Programs\Python\Launcher\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\msys64\mingw64\bin;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.89.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.1 
Workspace folder c:<HOMEPATH>\Desktop\EspExample\AOSYW\EMG 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\Dawid\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\Dawid\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Users\Dawid\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Users\Dawid\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Users\Dawid\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Users\Dawid\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Users\Dawid\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Users\Dawid\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Users\Dawid\.espressif\tools\cmake\3.24.0\bin;C:\Users\Dawid\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:\Users\Dawid\.espressif\tools\ninja\1.10.2;C:\Users\Dawid\.espressif\tools\idf-exe\1.0.3;C:\Users\Dawid\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Users\Dawid\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Users\Dawid\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\Users\Dawid\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Users\Dawid\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:\Users\Dawid\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM15
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-bridge.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\Dawid\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\Dawid\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:\Users\Dawid\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to C:\Users\Dawid\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to C:\Users\Dawid\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to C:\Users\Dawid\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to C:\Users\Dawid\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to C:\Users\Dawid\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:\Users\Dawid\.espressif\tools\cmake\3.24.0\bin: true
Access to C:\Users\Dawid\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin: true
Access to C:\Users\Dawid\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\Dawid\.espressif\tools\idf-exe\1.0.3: true
[... 136 lines cut ...]
    ],
    "version": 4
}
 
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF Doctor Command
--------------------------------------------------------------------------------------------------------------------------------------------

### Description

I Can't use correctly esp-idf even after formatting whole system to base state (no personal data). Extension don't execute ninja command

### Debug Message

```plain
*  Executing task: ninja  

ninja: error: loading 'build.ninja': The system cannot find the file specified.

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-05-20:
Could you share your extension log ?

Seems that there might be an issue with your python packages.

DawidGurdzinski2, 2024-05-20:
i can't paste extension info because it have  166,970 chars. so i upload file.
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/15376143/esp_idf_vsc_ext.log)

brianignacio5, 2024-05-21:
It seems that the extension can't access your esp-idf directory `C:\\Users\\Dawid\\esp\\esp-idf` as described in 

`{"message":"Cannot access filePath: C:\\Users\\Dawid\\esp\\esp-idf","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\Dawid\\esp\\esp-idf'\n\tat Object.accessSync (node:fs:264:3)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat ke (\\dist\\extension.js:2:1394261)\n\tat \\dist\\extension.js:2:1741020\n\tat \\dist\\extension.js:2:1741991\n\tat async \\dist\\extension.js:2:1739252","level":"error","timestamp":"2024-05-19T14:11:03.996Z"}`

DawidGurdzinski2, 2024-05-23:
I think the problem is that somehow in Win11 esp-idf extension take my online account username instead of my computer username and use is as a path.
Is there any fix for that?

brianignacio5, 2024-05-24:
Just update the `idf.espIdfPathWin` in your settings.json with the correct path or use the `ESP-IDF: Configure ESP-IDF extension` and use the correct path there.

brianignacio5, 2024-06-05:
Closing this issue due to lack of response here. Comment here if you need more assistance.

closed 2024-06-05 by brianignacio5