espressif/vscode-esp-idf-extension #1197: [Feature Request]: Add "Reconfigure" Button

viayie, 2024-05-22:
**Is your feature request related to a problem? Please describe.**
After making changes in the `SDK Configuration Editor(menuconfig)` and saving them, I have to trigger a full build to regenerate the `compile_command.json`. 
This is necessary for vscode to correctly expand macros, enabling accurate code navigation and inactive code region display. However, a full build is time-consuming and not always necessary.
[image]

**Describe the solution you'd like**
I would like a solution similar to other commands (`Build`, `Full Clean`, `Flash`, `Monitor`, etc). Specifically, I would like a `Reconfigure` button that corresponds to the `idf.py reconfigure` function. This would allow me to regenerate the `compile_command.json` without triggering a full build.
[image]

**Describe alternatives you've considered**
I've considered manually running `idf.py reconfigure` in the `Open ESP-IDF terminal`, but it is not as convenient as having a dedicated `Reconfigure` button directly in the editor.

**Additional context**
None.

Chris-R-U, 2024-05-22:
I find the problem too，thanks boy

brianignacio5, 2024-05-23:
Please try this feature with this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1526620398.zip)

PR #1198 (2024-05-27): add idf py reconfigure task
## Description

Add `ESP-IDF: Run idf.py reconfigure task` command to run `idf.py reconfigure` from extension.

Fixes #1197

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF: Run idf.py reconfigure task"
2. Execute command and See  make configure step being executed.
3. Observe results.

- Expected behaviour:

- Expected output:

## How has this been tested?

Manual testing running the command.

**Test Confi [... cut]
closed 2024-05-27 by brianignacio5