espressif/vscode-esp-idf-extension #1199: [vscode Registers/Peripherals change]:

ilangr-skyplug, 2024-05-23:
I was able to configure my vscode to enter debug when using ESP-PROG that is connected to ESP32-S3-DevKitC-1-N8 JTAG.
I can see watch, disassembly, memory, stack, registers & ESP-IDF: Peripherals.
For some reason I can't change the value of any of the registers or ESP peripherals in debug. 
All modern IDEs allows the programmer to change registers on the fly...
I was asked by your support to open a ticket for it

brianignacio5, 2024-06-05:
Have you try with the new Eclipse CDT Adapter in master ?

ilangr-skyplug, 2024-06-05:
Hi @brianignacio5 ,
No I haven't tried it. Is it a vscode extension?

brianignacio5, 2024-06-05:
So in [this Pull Request](https://github.com/espressif/vscode-esp-idf-extension/pull/1131) we added a new debug adapter from Eclipse CDT team here which should improve the debugging experience.

Please take a look with current GitHub master [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) and follow the update [debugging tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/debugging.md)

ilangr-skyplug, 2024-06-05:
Thanks @brianignacio5 . I'll look at it

closed 2024-06-26 by github-actions[bot]