espressif/vscode-esp-idf-extension #1201: groupadd fails

powerteqtpitman, 2024-05-23:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.89.1

### ESP-IDF version

latest

### Python version

3.10

### Doctor command output

I don't know how to get this

### Description

I am following this tutorial: https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/using-docker-container.md

When I get to the step of opening the dev container I get the following output and then a dialog that says it failed:

https://pastebin.com/p847zTaS

### Debug Message

```plain
I am not getting this far
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-05-24:
Open your project `.devcontainer` directory `Dockerfile` and change `ARG USER_UID=1000` to `ARG USER_UID=1050`

powerteqtpitman, 2024-05-28:
That fixed it. Thanks.

PR #1200 (2024-05-30): rm mount from devcontainer json template
## Description

Remove mount settings from devcontainer.json since they are not necessary. update docker file userid to avoid conflict with default user id.

Fixes #1190
Fixes #1201

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Create any ESP-IDF project from examples or New Project Wizard.
2. Open folder in container by running `Dev Containers: Open Folder in Container` c [... cut]
closed 2024-05-30 by brianignacio5