espressif/vscode-esp-idf-extension #1202: sdkconfig selected by menuconfig not as expected

coratron, 2024-05-28:
### OS

Linux

### Operating System version

Ubuntu 22.04

### Visual Studio Code version

1.89.1

### ESP-IDF version

5.1.2

### Python version

3.10.12

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.15.0-107-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.serverless/bin:<HOMEPATH>/flutter/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.pyenv/plugins/pyenv-virtualenv/shims:<HOMEPATH>/.pyenv/shims:<HOMEPATH>/.pyenv/bin:<HOMEPATH>/.linuxbrew/Homebrew/bin:<HOMEPATH>/bin:/usr/local/bin:<HOMEPATH>/.local/bin:<HOMEPATH>/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/opt/st/stm32cubeclt_1.12.0:/opt/st/stm32cubeclt_1.12.0/STM32CubeProgrammer/bin:/opt/st/stm32cubeclt_1.12.0/STLink-gdb-server/bin:/opt/st/stm32cubeclt_1.12.0/GNU-tools-for-STM32/bin:<HOMEPATH>/Documents/flutter/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.89.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/zsh 
ESP-IDF Extension version 1.7.1 
Workspace folder <HOMEPATH>/Documents/Repositories/flexi_plc_fw 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/v5.1.2/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.24.0/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.10.2:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.1_py3.10_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) /usr/bin/git
Notification Mode (idf.notificationMode) Output
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/cmake/3.24.0/bin: false
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/ninja/1.10.2: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320: true
[... 203 lines cut ...]
      "preBuild": "",
      "preFlash": "",
      "postBuild": "",
      "postFlash": ""
    }
  }
}
```
I am not 100% sure why I would specify _sdkconfigFilePath_ for menuconfig to then ignore this parameter. I don't know if I am missing the point or if it is a bug - or an intended feature in which case the menuconfig is rendered somewhat useless for multiple configuration projects.

Any ideas?

### Debug Message

```plain
No issues on firmware front - it's the build/configuration process that has a problem
```

### Other Steps to Reproduce

This can be reproduced on the multiple configuration example project of esp-idf.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-05-28:
There are 2 parts here to understand.

The `sdkconfig` is generated by the build system so this file is generated (or modified) every time you build. In case of multi config projects, the sdkconfig lives in the build directory to have separated results.

The case you want to use, have a common sdkconfig I will call `sdkconfig.common`, then sdkconfig.gen1 and/or sdkconfig.gen2 you want to define them in sdkconfig defaults, which are used to generate the end `sdkconfig` file used in each config build.

Something like:

```json
"buildDirectoryPath": "${workspaceFolder}/build_gen2",
"sdkconfigDefaults": [
  "sdkconfig.common",
  "sdkconfig.gen2"
],
"sdkconfigFilePath": "${workspaceFolder}/build_gen2/sdkconfig"
```

You can observe that in the [ESP-IDF multi config example](https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config#create-configuration-profile-files-via-filename) there are 2 files used in sdkconfig defaults `sdkconfig.prod_common;sdkconfig.prod1` where common has shared settings for both products and prod1 has product 1 specific settings.

coratron, 2024-05-28:
@brianignacio5 Thanks for your prompt reply. I think my point still stands though regarding menuconfig:

If I use the structure you have shared, menuconfig is still opening `${workspaceFolder}/build_gen2/sdkconfig`

The idea would be to be able to specifiy `sdkconfig.common` instead, or any sdkconfig the user decides to point to.

brianignacio5, 2024-05-28:
You can also try using `"sdkconfigFilePath": "${workspaceFolder}/sdkconfig.config2"` or other name.

coratron, 2024-05-28:
This is what I shared in my original configuration (circling back):
`"sdkconfigFilePath": "${workspaceFolder}/sdkconfig"`

Unfortunately it just opens the `sdkconfig` that is inside the build folder, no matter what I place in that field. When I make a change (and save to confirm it) that is the one that gets modified, and not the one specified in `sdkconfigFilePath`.

Some steps for further clarifications:
- Building the project `sdkconfig` is not in the build folder
- Open menuconfig change some parameter, save
- `sdkconfig` now appears in the build folder

Opening menuconfig only changes this `sdkconfig` , which means changing back and forth between configurations and deleting the build folders (or working with git essentially) will wipe these changes made to the specific `sdkconfig` of each configuration.

coratron, 2024-05-28:
> You can try using `"sdkconfigFilePath": "${workspaceFolder}/sdkconfig.config2"` or other name.

I have tried the following with no difference in outcome:
```

  "buildDirectoryPath": "${workspaceFolder}/build_gen1",
      "sdkconfigDefaults": [
        "sdkconfig.gen1",
        "sdkconfig"
      ],
      "sdkconfigFilePath": "${workspaceFolder}/sdkconfig.bananas"
```

brianignacio5, 2024-05-28:
In your CmakeLists.txt file, do you have something defined like this:

https://github.com/espressif/esp-idf/blob/master/examples/build_system/cmake/multi_config/CMakeLists.txt#L6

If so could you try commenting this line and rebuild first ?

coratron, 2024-05-28:
> In your CmakeLists.txt file, do you have something defined like this:
> 
> https://github.com/espressif/esp-idf/blob/master/examples/build_system/cmake/multi_config/CMakeLists.txt#L6
> 
> If so could you try commenting this line and rebuild first ?

That line is commented out. It's another option I had observed and tested just in case... No luck.

coratron, 2024-05-28:
Also, the way I have been working around this, is by running:
`idf.py menuconfig`

which still targets the sdkconfig file which I share for the various configurations - it is just a shame to not be able to do that with this vscode extension.

brianignacio5, 2024-05-28:
Although the extension does cache the values from the first load, you can use the `ESP-IDF: Dispose Current SDK Configuration Editor Server Process` to kill the backend process and load confserver again.

Can you try running the previous command, running SDK Configuration Editor again, and see the ESP-IDF output to see which sdkconfig value is used ?

coratron, 2024-05-28:
Still the same file in the build folder unfortunately:
```
Error: Failed to load from ****/build_gen1/sdkconfig: Could not open '****/build_gen1/sdkconfig' (ENOENT: No such file or directory). Check that the $srctree environment variable (unset or blank) is set correctly.

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
{"version": 2, "values": {}, "ranges": {}, "visible": {}, "error": ["Failed to load from ****/build_gen1/sdkconfig: Could not open '****/build_gen1/sdkconfig' (ENOENT: No such file or directory). Check that the $srctree environment variable (unset or blank) is set correctly."]}
```
which is a bit surprising because it does open it and then when I edit one parameter and save I get:

```
[SDK Configuration Editor]
{"version": 2, "set": { "BOOTLOADER_WDT_TIME_MS": 7000 }}

[SDK Configuration Editor]
Set BOOTLOADER_WDT_TIME_MS
{"version": 2, "values": {"BOOTLOADER_WDT_TIME_MS": 7000}, "ranges": {}, "visible": {}}

[SDK Configuration Editor]
{"version":2,"save":"****/build_gen1/sdkconfig"}
[SDK Configuration Editor]
Saving config to ****/build_gen1/sdkconfig...

[SDK Configuration Editor]
{"version": 2, "values": {}, "ranges": {}, "visible": {}}
```

coratron, 2024-06-24:
@brianignacio5  I will try to test this on my second machine this week and see if the same happens - I have been busy with other things and have not reviewed the issue since.

coratron, 2024-07-12:
@brianignacio5  I have finally tested this on another machine and I am seeing the same issues.

I have the two project configurations. No matter which one I select, when I open the "SDK Configuration Editor" by pressing the cog icon and make some changes and save them they always end up in the first configuration of the two.

```
{"version":2,"save":"xxxx/build_gen1/sdkconfig"}
Saving config to xxxx/build_gen1/sdkconfig...

{"version": 2, "values": {}, "ranges": {}, "visible": {}}
```

This should actually have been saved to /build_gen2 because that's the project configuration I was working with...

brianignacio5, 2024-07-17:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1710895911.zip) ?

brianignacio5, 2024-08-05:
Hi @coratron  Have you managed to see the PR with changes as you requested ?

PR #1252 (2024-10-11): fix support for multiple sdkconfig files
## Description

Fix support of multiple sdkconfig when using Project Configuration and multiple profiles.

Fixes #1202
Fixes #1265
Fixes #1283
Fixes #1299

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Using a project with multiple configurations like https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config run the `ESP-IDF: Open Project Configuration` and configure 2 profiles with different SDK Configuration Edito [... cut]
closed 2024-08-26 by github-actions[bot]