espressif/vscode-esp-idf-extension #1203: command:espIdf.getToolchainGdb undefined

marcinczenko, 2024-05-29:
### OS

MacOS

### Operating System version

macOS 14.5

### Visual Studio Code version

1.89.1

### ESP-IDF version

v5.4.0

### Python version

3.12.3

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 23.5.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /var/folders/ns/r3r4k_9j39q42bmfjh5s7pwh0000gn/T/frum_2333_1716516253621/bin:<HOMEPATH>/.pyenv/shims:<HOMEPATH>/micro:<HOMEPATH>/.bun/bin:<HOMEPATH>/.nvm/versions/node/v20.13.1/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/Apple/usr/bin:/Applications/Keybase.app/Contents/SharedSupport/bin:<HOMEPATH>/.cargo/bin:<HOMEPATH>/go/bin:<HOMEPATH>/.rvm/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.89.1 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.7.1 
Workspace folder <HOMEPATH>/code/embedded/native-esp-idf/example-native-esp-idf 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/code/embedded/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240305/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240305/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.4_py3.12_env/bin/python
Serial port (idf.port) /dev/cu.usbserial-0001
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/olimex-arm-usb-tiny-h.cfg,board/esp-wroom-32.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240305/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240305/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20240305: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
[... 192 lines cut ...]
[esp32.cpu0] Core was reset.
[esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
[esp32.cpu1] Core was reset.
[esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
[esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
[esp32.cpu0] Reset cause (3) - (Software core reset)
Register cache flushed.
Hardware assisted breakpoint 1 at 0x400d6370: file /Users/mczenko/code/embedded/native-esp-idf/example-native-esp-idf/main/gpio_example_main.c, line 82.
[esp32.cpu0] Target halted, PC=0x400D6370, debug_reason=00000001
Set GDB target to 'esp32.cpu0'
[esp32.cpu1] Target halted, PC=0x4008444A, debug_reason=00000000
[New Thread 1073412648]
[Remote target exited]
[New Thread 1073413336]
[New Thread 1073412992]
[New Thread 1073412272]
[New Thread 1073410900]
[Switching to Thread 1073412648]
Thread 2 "main" hit Temporary breakpoint 1, app_main () at /Users/mczenko/code/embedded/native-esp-idf/example-native-esp-idf/main/gpio_example_main.c:82
82	{
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TixoRebel, 2024-05-29:
I see this as well on linux

brianignacio5, 2024-05-30:
Are you using the extension from marketplace or the current master vsix ? We haven't release these changes to marketplace yet.

marcinczenko, 2024-05-31:
I am using the extension from the marketplace.

brianignacio5, 2024-05-31:
Please try the master latest installer in [here](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip).

We will release soon but we need to do extension testing before release new version.

brianignacio5, 2024-06-05:
Closing this issue since it is fixed in master and will be part of v1.8.0

closed 2024-06-05 by brianignacio5