espressif/vscode-esp-idf-extension #1208: ESP-Matter - Cannot Use Example ESP-Matter Projects

MindsGoneAgain, 2024-06-01:
### OS

Linux

### Operating System version

pop os 22.04

### Visual Studio Code version

1.89.1

### ESP-IDF version

5.2.1

### Python version

3.10.12

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 6.8.0-76060800daily20240311-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.89.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.7.1 
Workspace folder No workspace folder is open 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/v5.2.1/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) <HOMEPATH>/.espressif/esp-matter
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.2_py3.10_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) true
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
[... 88 lines cut ...]
After using the esp-idf extension to install esp-matter, I get an error (Cannot read properties of undefined (reading 'find')) and after installation when I select '>ESP-IDF: Show Examples Projects' and then 'Use Current ESP-Matter' it shows no examples.
[image]
[image]

Expected Behavior:
I should be able to install ESP-Matter without getting any errors and after installation I should be able to see ESP-Matter examples and create projects from them.

### Debug Message

```plain
Cannot read properties of undefined (reading 'find')
```

### Other Steps to Reproduce

1. Install VS Code and ESP-IDF extension following instructions here: https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md
2. Configure ESP-IDF extension using express setup
3. Install ESP-Matter using the 'ESP-IDF: Install ESP-Matter' command
4. Try to open the ESP-Matte examples

Additionally, I tried creating an example project by copying an example from the examples folder, but it threw errors. I have no trouble using the regular ESP-IDF examples.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-06-03:
Found the issue and fixed in #1207 The issue was that when no folder is open the matter bootstrap task is not being executed. If you open a folder then try to run `ESP-IDF: Install ESP-Matter` again it should work.

aarbmx6s, 2024-06-13:
I find that on my Mac, I cannot see examples because the ESP-IDF extension cannot find the `esp-matter/examples/common/blemesh_platform/platform/ESP32_custom/WiFiDnssdImpl.cpp` file. I see the file in the explorer but cannot open it. Actually, the file is not a file but a symlink.

In the example, I installed esp-matter into the project folder. However, it behaves the same way if it is installed globally.

[image]

[image]

brianignacio5, 2024-06-19:
It should not be related to examples being shown @aarbmx6s As long as idf.espMatterPath is set you should see the examples.

This is probably because a bug in reading settings. Should be fixed with #1207 you can test it with this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1612784298.zip)

PR #1207 (2024-06-28): v1.8.0 release
## Description

Add changes for release v1.8.0

Fix #1227 
Fix #1208 
Fix #1210
Fix #1213 
Fix #1217

## Type of change

- This change requires a documentation update
- Bug fixes 

## Steps to test this pull request

1. Test `ESP-IDF: Install ESP-Matter` without any folder opened. Matter should be installed without issues and `idf.espMatterPath` should be set and examples shown.
2. Create a new ESP-IDF project and launch `Start OpenOCD`, the settings from workspace folder should be used without updating settings.json.
A Complete test of the extension is required.

## How has thi [... cut]
closed 2024-06-28 by brianignacio5