espressif/vscode-esp-idf-extension #1209: Command "ESP-IDF: New Project" does not work (IDFGH-12937)

DemTerritory, 2024-06-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF v1.7.1 (VSCode)
It is impossible to create a new project using the "ESP-IDF: New Project" command from the Command Palette. "Create project using template..." button doesn't work without any messages or errors

[image]

brianignacio5, 2024-06-03:
Please share the [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension/blob/master/README.md#Troubleshooting)

DemTerritory, 2024-06-03:
I'm sorry, I was wrong, a new project is being created. But it is created literally in a split second, and upon completion a pop-up message: “The project has been created, open the project in a new window?” doesn't appear. The message was actually created, but remained in history under the bell. And when I tried to create the same project again, the message “the project already exists, should I overwrite the contents?” appeared, but it also remained not open in the history under the bell
[image]

brianignacio5, 2024-06-05:
Closing this since there is no issue then.

closed 2024-06-05 by brianignacio5