espressif/vscode-esp-idf-extension #1210: Start OpenOCD ignores configuration in settings.json

veeman, 2024-06-04:
### OS

Windows

### Operating System version

Windows 10, WSL2

### Visual Studio Code version

1.89.1

### ESP-IDF version

5.1.1

### Python version

3.8.10

### Doctor command output

```

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.10.102.1-microsoft-standard-WSL2 
System environment variable IDF_PYTHON_ENV_PATH 
 /opt/esp/python_env/idf5.1_py3.8_env 
System environment variable PATH 
 /vscode/vscode-server/bin/linux-x64/dc96b837cf6bb4af9cd736aa3af08cf8279f7685/bin/remote-cli:/opt/esp/idf/components/espcoredump:/opt/esp/idf/components/partition_table:/opt/esp/idf/components/app_update:/opt/esp/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:/opt/esp/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/opt/esp/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/opt/esp/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin:/opt/esp/python_env/idf5.1_py3.8_env/bin:/opt/esp/idf/tools:/opt/esp/tools/cmake/3.27.4/bin:/opt/esp-clang/bin:/opt/qemu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.89.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.7.1 
Workspace folder /workspaces/FW-AB2000-ESP32 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /opt/esp/idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars)
Virtual env Python Path (idf.pythonBinPath) /opt/esp/python_env/idf5.1_py3.8_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) interface/jlink.cfg,target/esp32.cfg,jtag-config.cfg
ESP-IDF Tools Path (idf.toolsPath) /opt/esp
Git Path (idf.gitPath) /usr/bin/git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.25.1
ESP-IDF version 5.1.1
Python version 3.8.10
Python's pip version 23.2.1
-------------------------------------------------- Project configuration settings ----------------------------------------------------------
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitarray version: 2.8.0
bitstring version: 4.1.0
CacheControl version: 0.13.1
certifi version: 2023.7.22
cffi version: 1.15.1
charset-normalizer version: 3.2.0
click version: 8.0.4
colorama version: 0.4.6
construct version: 2.10.68
contextlib2 version: 21.6.0
cryptography version: 39.0.2
ecdsa version: 0.18.0
esp-coredump version: 1.6.0
esp-debug-backend version: 1.0.3
esp-idf-kconfig version: 1.2.0
esp-idf-monitor version: 1.1.3
esp-idf-size version: 0.4.2
esptool version: 4.7.dev1
filelock version: 3.12.2
freertos-gdb version: 1.0.2
idf-component-manager version: 1.3.2
idna version: 3.4
kconfiglib version: 14.1.0
msgpack version: 1.0.5
packaging version: 23.1
pip version: 23.2.1
pkg_resources version: 0.0.0
pycparser version: 2.21
pyelftools version: 0.29
pygdbmi version: 0.9.0.2
pyparsing version: 3.0.9
pyserial version: 3.5
PyYAML version: 6.0.1
reedsolo version: 1.7.0
requests version: 2.32.2
requests-file version: 1.5.1
requests-toolbelt version: 1.0.0
schema version: 0.7.5
setuptools version: 70.0.0
six version: 1.16.0
tqdm version: 4.66.1
urllib3 version: 1.26.16
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Python requirements are satisfied.
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Python requirements are satisfied.
---------------------------------------------------- Visual Studio Code launch.json --------------------------------------------------------
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ESP-IDF: Debug",
            "type": "espidf",
            "request": "launch",
            "mode": "auto"
        }
    ]
} 
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF Doctor Command
--------------------------------------------------------------------------------------------------------------------------------------------
```

### Description

When attempting to start the OpenOCD server by running `Start OpenOCD` within a folder, the server does not start with the configurations defined in the settings.json file. To resolve this issue, we must first open the `settings.json` file, modify it, and then save it. Only after this step the server start with the configurations specified in the `idf.openOcdConfigs` config.

Associated `settings.json`:
```json
{
    "idf.adapterTargetName": "esp32",
    "idf.openOcdConfigs": [
        "interface/jlink.cfg",
        "target/esp32.cfg",
        "jtag-config.cfg"
    ],
    "idf.flashType": "JTAG",
    "idf.monitorBaudRate": "115200",
}
```

**Expected Behavior**

The OpenOCD server should start with the configurations defined in the `settings.json` file upon running the `Start OpenOCD` command without needing to modify and save the `settings.json` file first.

**Actual Behavior**

The OpenOCD server does not start with the configurations defined in the `settings.json` file until the file is modified and saved.

### Debug Message

```plain
Actually it looks like it is invoked OpenOCD with the default settings:

/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/openocd

[OpenOCD]
Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
adapter speed: 20000 kHz
WARNING: boards/esp32-wrover.cfg is deprecated, and may be removed in a future release.
         If your board is ESP32-WROVER-KIT, use board/esp32-wrover-kit-1.8v.cfg instead.

[OpenOCD]
Info : Listening on port 6666 for tcl connections

[OpenOCD]
Info : Listening on port 4444 for telnet connections

[OpenOCD]
❌ Error: unable to open ftdi device with description '*', serial '*' at bus location '*'

[OpenOCD]
❌ /opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:4: Error: 
at file "/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 4

[OpenOCD]
For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
[OpenOCD]
OpenOCD Exit with non-zero error code 1
[OpenOCD]
[Stopped] : OpenOCD Server
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-06-05:
You can try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1569597160.zip) to see if the issue is fixed.

veeman, 2024-06-05:
I've tried the new version provided. The same behavior persists.

I have narrowed down the issue and found that it only occurs when launching from within a `*.code-workspace` file.

brianignacio5, 2024-06-05:
When you use `*.code-workspace`, do you select the current workspace folder with `ESP-IDF: Pick a Workspace Folder ` as described in [multiple projects documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/MULTI_PROJECTS.md) ?

veeman, 2024-06-05:
Currently i have only one folder in my workspace. I tried also `ESP-IDF: Pick a Workspace Folder`, but this does not help either.

Assocaited workspace file:
```json
{
  "folders": [
    {
      "name": "FW-AB2000-ESP32",
      "path": "."
    }
  ],
  "settings": {
    // other non-idf settings
  },
  "extensions": {
    "recommendations": [
      "cschlosser.doxdocgen",
      "espressif.esp-idf-extension",
      "gruntfuggly.todo-tree",
      "llvm-vs-code-extensions.vscode-clangd"
    ]
  }
}
```

And my devcontainer.json:
```json
{
  "name": "ESP-IDF",
  "build": {
    "dockerfile": "Dockerfile"
  },
  "customizations": {
    "vscode": {
      // Pre-configure ESP-IDF extension
      "settings": {
        "terminal.integrated.defaultProfile.linux": "bash",
        "idf.showOnboardingOnInit": false,
        "idf.toolsPath": "/opt/esp",
        "idf.toolsPathWin": "",
        "idf.espIdfPath": "/opt/esp/idf",
        "idf.pythonBinPath": "/opt/esp/python_env/idf5.1_py3.8_env/bin/python",
        "idf.customExtraPaths": "",
        "idf.port": "/dev/ttyUSB0"
      }
    }
  },
  "userEnvProbe": "loginInteractiveShell",
  // Make ttyUSB and raw USB devices accessible within the container
  "mounts": [
    // Access to /dev
    "type=bind,source=/dev,target=/dev"
  ],
  "runArgs": [
    // USB serial adapters
    "--device-cgroup-rule",
    "c 188:* rmw",
    // raw USB devices
    "--device-cgroup-rule",
    "c 189:* rmw"
  ]
}
```

PR #1207 (2024-06-28): v1.8.0 release
## Description

Add changes for release v1.8.0

Fix #1227 
Fix #1208 
Fix #1210
Fix #1213 
Fix #1217

## Type of change

- This change requires a documentation update
- Bug fixes 

## Steps to test this pull request

1. Test `ESP-IDF: Install ESP-Matter` without any folder opened. Matter should be installed without issues and `idf.espMatterPath` should be set and examples shown.
2. Create a new ESP-IDF project and launch `Start OpenOCD`, the settings from workspace folder should be used without updating settings.json.
A Complete test of the extension is required.

## How has thi [... cut]
closed 2024-06-26 by github-actions[bot]