espressif/vscode-esp-idf-extension #1214: Update Documentation for Docker on Mac with USB/Serial passthrough

emanuelbalea, 2024-06-13:
### OS

MacOS

### Operating System version

14

### Visual Studio Code version

1.7.1

### ESP-IDF version

5.3

### Python version

3.10.12

### Doctor command output

Not applicable

### Extension

Not Applicable

### Description

As a developer I have been trying to maintain a consistent and portable environment between the other developers in the team.
I currently use the docker image for the build server and for dev using Docker via WSL2 on windows where using usbipd works as described at [docs/tutorial/using-docker-container.md](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/using-docker-container.md). However while it mentions mac os near the start there is no other mention of it further along.

Since a lot of our developers now use macbooks for combined ios/esp32 dev, we need to be able to do the same under mac os. Currently build works in docker on mac. However I have not been able to set up usb passthrough using the devcontainer.json file.

Could we please get this documented by someone who understands how devcontainers work? I have not been able to find consistent information about this using google searches as they reference older docker issues or framework versions.

### Debug Message

```plain
Not Applicable
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-06-21:
Hi @emanuelbalea sorry for late reply, I will test using docker container in MacOS but I believe the missing piece is configuring the docker container to see the serial ports. On Windows we have to do usbipd and since many users requested it was added, after we finish  v1.8.0 release testing we will update documentation regarding macOS docker container use.

Apologies for the delay.

emanuelbalea, 2024-07-07:
Just bumping this issue to stop to stale bot as not sure if there is an internal tracking task for this or not.

brianignacio5, 2024-07-11:
Unfortunately USB Passthrough is not yet possible with Docker on macOS as described in this [github issue](https://github.com/moby/hyperkit/issues/149).

You can try to use a remote serial port with esptool.py RFC server as shown in [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html?highlight=docker#using-remote-serial-port)

closed 2024-08-20 by brianignacio5

TheKing349, 2025-11-20:
The virtualization framework does allow for serial ports and usbControllers. Is this something that could be added? The hypervisor framework does have a Boolean to allow for USB devices to connect as well

PR #1231 (2024-08-20): Prepare documentation for Espressif docs
## Description

Add RST files for Espressif documentation release. After finishing this PR we will have a new one for Chinese translation of these files.

Fix #1214 

## Type of change

- This change requires a documentation update

## Steps to test this pull request

Documentation change which will be available as preview documentation.

- Expected behaviour:

- Expected output:

## How has this been tested?

Open preview documentation here To-DO add preview docs link

## Checklist
- [ ] PR Self Reviewed
- [ ] Applied Code formatting
- [ ] Added Docume [... cut]