espressif/vscode-esp-idf-extension #1217: Create project using template option Doesn't work

5ami, 2024-06-16:
### OS

Windows

### Operating System version

windows 11

### Visual Studio Code version

1.89.1

### ESP-IDF version

5.1

### Python version

3.11.2

### Doctor command output

NONE

### Description

Create project using template option Doesn't work 
[image]

### Debug Message

```plain
NONE
```

### Other Steps to Reproduce

TRY to Create project using template option

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-06-17:
What do mean it doesn't work ? You should see a pick a folder window and then a notification that is created (if you have notifications enabled) Also why you don't provide extension log and doctor command if specified in the issue template? Errors are printed there.

5ami, 2024-06-17:
Hi brianignacio, the reason I didn't provide anything is because the button is not clickable. Clicking the button results in Basically Nothing.

brianignacio5, 2024-06-17:
I meant if you could share the troubleshooting information from:

1. In Visual Studio Code select menu `View` then click `Command Palette...` and type `ESP-IDF: Doctor Command` to generate a report of your environment configuration and it will be copied in your clipboard to paste anywhere.

3. Check extension log file which can be obtained from:
Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

5ami, 2024-06-17:
[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/15863860/esp_idf_vsc_ext.log)
[report.txt](https://github.com/user-attachments/files/15863861/report.txt)

5ami, 2024-06-17:
@brianignacio5  could you reproduce the issue ? 
 I think the problem is that the button lacks back-end call  because I could reproduce this issue in three different computers. So I doubt that it is something related to an environment issue, its rather a problem with the extension itself.

brianignacio5, 2024-06-17:
I can't reproduce the issue, when you click the button a system open folder window should appear right away.

Could you uninstall the extension and try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) ?

5ami, 2024-06-17:
> I can't reproduce the issue, when you click the button a system open folder window should appear right away.
> 
> Could you uninstall the extension and try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) ?

BTW, the vsix behaves the same

brianignacio5, 2024-06-20:
I think I found the solution and added the changes here: [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1617122045.zip) Please take a look if you have some time.

5ami, 2024-06-20:
@brianignacio5  Thanks it seems to work, just for my curiosity what was the issue ? 
Also I have to test the debugger as Last time i tried the VSIX installer the debugging wasn't working correctly , so I have to double check that

5ami, 2024-06-20:
@brianignacio5  I tested the debugger with vsix installer, seems to work  so you can close the issue.

PR #1207 (2024-06-28): v1.8.0 release
## Description

Add changes for release v1.8.0

Fix #1227 
Fix #1208 
Fix #1210
Fix #1213 
Fix #1217

## Type of change

- This change requires a documentation update
- Bug fixes 

## Steps to test this pull request

1. Test `ESP-IDF: Install ESP-Matter` without any folder opened. Matter should be installed without issues and `idf.espMatterPath` should be set and examples shown.
2. Create a new ESP-IDF project and launch `Start OpenOCD`, the settings from workspace folder should be used without updating settings.json.
A Complete test of the extension is required.

## How has thi [... cut]
closed 2024-06-28 by brianignacio5