espressif/vscode-esp-idf-extension #1219: Failed to set device target in VSCode extension of ESP-IDF

OW1TY2, 2024-06-17:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.90.1

### ESP-IDF version

5.1.4

### Python version

3.12.4

### Doctor command output

VCC-GND ESP32-S3-WROOM-1U-N16R8 dev board, connecting a LCD module and a SHTC3 sensor.
[esp-idf doctor.txt](https://github.com/user-attachments/files/15864801/esp-idf.doctor.txt)
[image]

### Extension

[esp_idf_vsc_ext.log](https://github.com/user-attachments/files/15864747/esp_idf_vsc_ext.log)

### Description

When I try to configure the device target, the terminal pop up some error message below and I cannot configure to my target ESP32-S3

### Debug Message

```plain
The following Python requirements are not satisfied:

Requirement 'pyparsing<3.1,>=3.0.3' was not met. Installed version: 3.1.2

Requirement 'cryptography<40,>=2.1.4' was not met. Installed version: 41.0.7

Requirement 'pyelftools<=0.29' was not met. Installed version: 0.30

To install the missing packages, please run "install.bat"

Diagnostic information:

    IDF_PYTHON_ENV_PATH: C:/Espressif/python_env/idf5.1_py3.11_env

    Python interpreter used: C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe

    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH

    PATH: C:\Espressif\frameworks\esp-idf-v5.1.4\components\esptool_py\esptool;C:\Espressif\frameworks\esp-idf-v5.1.4\components\espcoredump;C:\Espressif\frameworks\esp-idf-v5.1.4\components\partition_table;C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\tools\esp-rom-elfs\20230320;C:/Espressif/python_env/idf5.1_py3.11_env/Scripts;C:\Espressif\frameworks\esp-idf-v5.1.4\tools;C:/Espressif/tools/idf-git/2.44.0/cmd;c:\Users\cheun\.vscode\extensions\cl.eide-3.18.0\res\tools\win32\unify_builder;C:\Users\cheun\.eide\bin\builder\utils;C:\Users\cheun\.eide\bin\scripts;c:\Users\cheun\.vscode\extensions\cl.eide-3.18.0\res\tools\win32\7z;c:\Users\cheun\.vscode\extensions\cl.eide-3.18.0\res\tools\win32\utils;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;.;C:\Users\cheun\.eide\bin\builder\msys\bin;C:\Users\cheun\AppData\Local\Programs\Python\Python312\Scripts\;C:\Users\cheun\AppData\Local\Programs\Python\Python312\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Oracle\Java\java8path;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\dotnet\;C:\Program Files\dotnet\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\msys64\ucrt64\bin;C:\Program Files\Wolfram Research\WolframScript\;D:\MATLAB 2023b\runtime\win64;D:\MATLAB 2023b\bin;C:\Program Files\Calibre2\;C:\Program Files\Git\cmd;C:\Users\cheun\AppData\Local\Programs\Python\Python312\Scripts\;C:\Users\cheun\AppData\Local\Programs\Python\Python312\;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:\Users\cheun\AppData\Local\Microsoft\WindowsApps;C:\Users\cheun\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\cheun\.dotnet\tools;C:\Program Files\JetBrains\CLion 2023.3.4\bin;;C:\Users\cheun\AppData\Local\GitHubDesktop\bin;C:\intelFPGA\18.1\modelsim_ase\win32aloem;C:\intelFPGA\23.1std\questa_fse\win64

Constraint file: C:\Espressif\espidf.constraints.v5.1.txt

Requirement files:

 - C:/Espressif/frameworks/esp-idf-v5.1.4/tools\requirements\requirements.core.txt

Python being checked: C:/Espressif/python_env/idf5.1_py3.11_env\Scripts\python.exe

ESP-IDF v5.1.4-dirty

non zero exit code 1

The following Python requirements are not satisfied:

Requirement 'pyparsing<3.1,>=3.0.3' was not met. Installed version: 3.1.2

Requirement 'cryptography<40,>=2.1.4' was not met. Installed version: 41.0.7

Requirement 'pyelftools<=0.29' was not met. Installed version: 0.30

To install the missing packages, please run "install.bat"

Diagnostic information:

    IDF_PYTHON_ENV_PATH: C:/Espressif/python_env/idf5.1_py3.11_env

    Python interpreter used: C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe

    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH

    PATH: C:\Espressif\frameworks\esp-idf-v5.1.4\components\esptool_py\esptool;C:\Espressif\frameworks\esp-idf-v5.1.4\components\espcoredump;C:\Espressif\frameworks\esp-idf-v5.1.4\components\partition_table;C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\tools\esp-rom-elfs\20230320;C:/Espressif/python_env/idf5.1_py3.11_env/Scripts;C:\Espressif\frameworks\esp-idf-v5.1.4\tools;C:/Espressif/tools/idf-git/2.44.0/cmd;c:\Users\cheun\.vscode\extensions\cl.eide-3.18.0\res\tools\win32\unify_builder;C:\Users\cheun\.eide\bin\builder\utils;C:\Users\cheun\.eide\bin\scripts;c:\Users\cheun\.vscode\extensions\cl.eide-3.18.0\res\tools\win32\7z;c:\Users\cheun\.vscode\extensions\cl.eide-3.18.0\res\tools\win32\utils;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;.;C:\Users\cheun\.eide\bin\builder\msys\bin;C:\Users\cheun\AppData\Local\Programs\Python\Python312\Scripts\;C:\Users\cheun\AppData\Local\Programs\Python\Python312\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Oracle\Java\java8path;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\dotnet\;C:\Program Files\dotnet\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\msys64\ucrt64\bin;C:\Program Files\Wolfram Research\WolframScript\;D:\MATLAB 2023b\runtime\win64;D:\MATLAB 2023b\bin;C:\Program Files\Calibre2\;C:\Program Files\Git\cmd;C:\Users\cheun\AppData\Local\Programs\Python\Python312\Scripts\;C:\Users\cheun\AppData\Local\Programs\Python\Python312\;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:\Users\cheun\AppData\Local\Microsoft\WindowsApps;C:\Users\cheun\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\cheun\.dotnet\tools;C:\Program Files\JetBrains\CLion 2023.3.4\bin;;C:\Users\cheun\AppData\Local\GitHubDesktop\bin;C:\intelFPGA\18.1\modelsim_ase\win32aloem;C:\intelFPGA\23.1std\questa_fse\win64

Constraint file: C:\Espressif\espidf.constraints.v5.1.txt

Requirement files:

 - C:/Espressif/frameworks/esp-idf-v5.1.4/tools\requirements\requirements.core.txt

Python being checked: C:/Espressif/python_env/idf5.1_py3.11_env\Scripts\python.exe

ESP-IDF v5.1.4-dirty
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-06-17:
Hi there, it seems there is an issue with your virtual env packages missing some packages.

I would recommend to delete `C:\Espressif\python_env\idf5.1_py3.11_env` and try setup again.

OW1TY2, 2024-06-17:
> Hi there, it seems there is an issue with your virtual env packages missing some packages.
> 
> I would recommend to delete `C:\Espressif\python_env\idf5.1_py3.11_env` and try setup again.

Now the error message ""C:\Espressif\tools\idf-python\3.11.2\python.exe -m pip" is not valid. (ERROR_INVALID_PIP)" pop up when I tried to install from local IDF and download from Github

OW1TY2, 2024-06-17:
yet i have already installed pip

brianignacio5, 2024-06-18:
Seems that the embed python from local IDF doesn't have pip for some reason. Pip might exist in a different location in your setup. Could you make sure to either delete C:\Espressif\tools\idf-python (to redownload it) or install pip in it ?

OW1TY2, 2024-06-18:
`Error: spawn C:/Espressif/python_env/idf5.2_py3.11_env/Scripts/python.exe ENOENT at Process.ChildProcess._handle.onexit (node:internal/child_process:286:19) at onErrorNT (node:internal/child_process:484:16) at processTicksAndRejections (node:internal/process/task_queues:82:21) { errno: -4058, code: 'ENOENT', syscall: 'spawn C:/Espressif/python_env/idf5.2_py3.11_env/Scripts/python.exe', path: 'C:/Espressif/python_env/idf5.2_py3.11_env/Scripts/python.exe', spawnargs: [ 'C:\\Espressif\\frameworks\\esp-idf-v5.2.1\\tools\\idf.py', '--list-targets' ] }`

Now this error pop up

brianignacio5, 2024-06-18:
Could you get the latest doctor command output ?

OW1TY2, 2024-06-18:
`---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 c:<HOMEPATH>\.vscode\extensions\cl.eide-3.18.0\res\tools\win32\unify_builder;C:<HOMEPATH>\.eide\bin\builder\utils;C:<HOMEPATH>\.eide\bin\scripts;c:<HOMEPATH>\.vscode\extensions\cl.eide-3.18.0\res\tools\win32\7z;c:<HOMEPATH>\.vscode\extensions\cl.eide-3.18.0\res\tools\win32\utils;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;.;C:<HOMEPATH>\.eide\bin\builder\msys\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Oracle\Java\java8path;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\dotnet\;C:\Program Files\dotnet\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\msys64\ucrt64\bin;C:\Program Files\Wolfram Research\WolframScript\;D:\MATLAB 2023b\runtime\win64;D:\MATLAB 2023b\bin;C:\Program Files\Calibre2\;C:\Program Files\Git\cmd;C:\Espressif\tools\idf-python\3.11.2\;C:\Espressif\tools\idf-python\3.11.2\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312\;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\.dotnet\tools;C:\Program Files\JetBrains\CLion 2023.3.4\bin;;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:\intelFPGA\18.1\modelsim_ase\win32aloem;C:\intelFPGA\23.1std\questa_fse\win64 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.90.1 
Visual Studio Code language zh-cn 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.7.1 
Workspace folder c:<HOMEPATH>\AppData\Local\Arduino15\staging\libraries\TFT_eSPI-2.5.43\examples\Test and diagnostics\Read_User_Setup 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/frameworks/esp-idf-v5.2.1/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.11.1;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:/Espressif/python_env/idf5.2_py3.11_env/Scripts/python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif
Git Path (idf.gitPath) C:\Espressif\tools\idf-git\2.39.2\cmd\git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin: true
Access to C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:\Espressif\tools\cmake\3.24.0\bin: true
Access to C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to C:\Espressif\tools\ninja\1.11.1: true
Access to C:\Espressif\tools\idf-exe\1.0.3: true
[... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF Doctor Command
--------------------------------------------------------------------------------------------------------------------------------------------
`

brianignacio5, 2024-06-18:
You can see that `Access to Virtual env Python Path (idf.pythonBinPath) false` meaning that `C:/Espressif/python_env/idf5.2_py3.11_env/Scripts/python.exe` might not exist or is not accessible.

OW1TY2, 2024-06-18:
problem resolved after reinstall, thank you so much!

closed 2024-06-18 by OW1TY2