espressif/vscode-esp-idf-extension #1223: [Feature Request]:

OrhanYigitDurmaz, 2024-06-18:
**Is your feature request related to a problem? Please describe.**
The serial usb access is fixed on ubuntu with adding user to the dialout group.

**Describe the solution you'd like**
This group is called uucp in arch linux. Maybe add this warning for arch users when upload fails?
`sudo usermod -a -G uucp $USER`

brianignacio5, 2024-06-21:
Hi @OrhanYigitDurmaz Can you specify where is the issue you are referring being produced? I don't believe we have any dialout related documentation in this repository, only in the docker file template which uses Ubuntu (from esp-idf docker image).

OrhanYigitDurmaz, 2024-06-21:
> Hi @OrhanYigitDurmaz Can you specify where is the issue you are referring being produced? I don't believe we have any dialout related documentation in this repository, only in the docker file template which uses Ubuntu (from esp-idf docker image).

I'm sorry for my mistake. Looks like this is not with the vscode extension, but rather from the esptool itself. I will close this issue.

closed 2024-06-21 by OrhanYigitDurmaz