espressif/vscode-esp-idf-extension #1230: 使用监视器打开串口，始终提示：“文件名、目录名或卷标语法不正确”，1.7版本的时候是可用的

caidamao, 2024-06-28:
### OS

Windows

### Operating System version

win10

### Visual Studio Code version

up to date 

### ESP-IDF version

5.2.2

### Python version

3.11

### Doctor command output

Microsoft Windows [版本 10.0.19045.4529]
(c) Microsoft Corporation。保留所有权利。

C:\Users\QDM\Desktop\xxxx>set IDF_PATH='d:\IDF\IDF5.2.2\v5.2.2\esp-idf'

C:\Users\QDM\Desktop\C1M_PLUS>'d:\IDF\IDF5.2.2\python_env\idf5.2_py3.11_env\Scripts\python.exe' 'd:\IDF\IDF5.2.2\v5.2.2\esp-idf\tools\idf_monitor.py' -p COM22 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 'c:\Users\QDM\Desktop\xxxx\build\xxxx.elf'
文件名、目录名或卷标语法不正确。
为了保密 真实项目名按xxxx替换

### Description

更新1.8.0后，突然串口工具不能使用了。1.7.1是可以的，我的安装路径不在C盘，而是d:\IDF\IDF5.2.2\v5.2.2\esp-idf

### Debug Message

```plain
无
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-06-28:
Please share the troubleshooting information as required in the issue template.

From what you have shared I can't see where is the issue or how is the issue produced.

caidamao, 2024-06-28:
https://github.com/espressif/vscode-esp-idf-extension/assets/82438676/a831f8a6-70e8-4c46-8082-f3f824f6b5bc

这是故障视频，我编译了一个纯净的“hello world”例程。Here's the glitch video, and I compiled a clean "hello world" routine. 出现了这个提示

This prompt appears

caidamao, 2024-06-28:
https://github.com/espressif/vscode-esp-idf-extension/assets/82438676/21b5d42a-5ade-46e5-b8e6-dc86d2d71658
这是故障视频，我编译了一个纯净的“hello world”例程。Here's the glitch video, and I compiled a clean "hello world" routine. 出现了这个提示
This prompt appears
@brianignacio5

caidamao, 2024-06-28:
我尝试了使用idf.py monotor直接开启，是可行的，但今天下午更新到1.8后，快捷按钮就出现了此问题，在上午未更新的版本1.7.1内，仍旧可用
I tried using idf.py monitor to start directly, and it worked. But after updating to version 1.8 this afternoon, this problem appeared on the quick button. In the previous version 1.7.1 before the update in the morning, it was still working.

caidamao, 2024-06-28:
当我执行这段指令{'d:\IDF\IDF5.2.2\python_env\idf5.2_py3.11_env\Scripts\python.exe'}的时候显示错误，但是当我执行{d:\IDF\IDF5.2.2\python_env\idf5.2_py3.11_env\Scripts\python.exe}时成功了

When I execute this command {'d:\IDF\IDF5.2.2\python_env\idf5.2_py3.11_env\Scripts\python.exe'}, it shows an error, but when I execute d:\IDF\IDF5.2.2\python_env\idf5.2_py3.11_env\Scripts\python.exe directly, it succeeds.

caidamao, 2024-06-28:
[image]
Microsoft Windows [版本 10.0.19045.4529]
(c) Microsoft Corporation。保留所有权利。

C:\Users\QDM\Desktop\hello_world>'d:\IDF\IDF5.2.2\python_env\idf5.2_py3.11_env\Scripts\python.exe'
文件名、目录名或卷标语法不正确。 

C:\Users\QDM\Desktop\hello_world>d:\IDF\IDF5.2.2\python_env\idf5.2_py3.11_env\Scripts\python.exe
Python 3.11.2 (tags/v3.11.2:878ead1, Feb  7 2023, 16:38:35) [MSC v.1934 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>

caidamao, 2024-06-28:
我将指令修正为{d:\IDF\IDF5.2.2\python_env\idf5.2_py3.11_env\Scripts\python.exe d:\IDF\IDF5.2.2\v5.2.2\esp-idf\tools\idf_monitor.py -p COM22 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 c:\Users\QDM\Desktop\hello_world\build\hello_world.elf}后，可以正常激活串口窗口
After I revised the command to {d:\IDF\IDF5.2.2\python_env\idf5.2_py3.11_env\Scripts\python.exe d:\IDF\IDF5.2.2\v5.2.2\esp-idf\tools\idf_monitor.py -p COM22 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 c:\Users\QDM\Desktop\hello_world\build\hello_world.elf}, the serial port window can be activated normally.
似乎问题是因为‘’导致的
It seems that the problem is caused by the double quotes ' '.

caidamao, 2024-06-28:
Here is the successful log:

Microsoft Windows [版本 10.0.19045.4529]
(c) Microsoft Corporation。保留所有权利。

C:\Users\QDM\Desktop\hello_world>d:\IDF\IDF5.2.2\python_env\idf5.2_py3.11_env\Scripts\python.exe d:\IDF\IDF5.2.2\v5.2.2\esp-idf\tools\idf_monitor.py -p COM22 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 c:\Users\QDM\Desktop\hello_world\build\hello_world.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM22 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM22 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6328 
load:0x40078000,len:17952
load:0x40080400,len:4    
0x40080400: _init at ??:?

ho 8 tail 4 room 4       
load:0x40080404,len:3984 
entry 0x40080660
I (277) cpu_start: Multicore app
I (285) cpu_start: Pro cpu start user code
I (285) cpu_start: cpu freq: 240000000 Hz 
I (285) cpu_start: Application information:       
I (288) cpu_start: Project name:     C1M_PLUS     
I (293) cpu_start: App version:      abb9662-dirty
I (299) cpu_start: Compile time:     Jun 28 2024 17:33:12
I (305) cpu_start: ELF file SHA256:  6bf8a2ff7c2506c3... 
Warning: checksum mismatch between flashed and built applications. Checksum of built application is f2821baa4caacc8e2b3c61679e01f26d4ba6956804d836efc7a60487518ea375
I (311) cpu_start: ESP-IDF:          v5.2.2
I (316) cpu_start: Min chip rev:     v0.0
I (320) cpu_start: Max chip rev:     v3.99 
I (325) cpu_start: Chip rev:         v1.1
I (330) heap_init: Initializing. RAM available for dynamic allocation:
I (337) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (343) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (349) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (355) heap_init: At 3FFCC0C8 len 00013F38 (79 KiB): DRAM
I (361) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (368) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (374) heap_init: At 4009B738 len 000048C8 (18 KiB): IRAM
I (381) spi_flash: detected chip: generic
I (385) spi_flash: flash io: qio
I (389) coexist: coex firmware version: d96c1e51f
I (395) main_task: Started on CPU0
I (399) main_task: Calling app_main()
E (420) USER_NVS_lib: wifi_info nvs句柄开启失败 err_code=ESP_ERR_NVS_NOT_FOUND
E (420) USER_NVS_lib: wifi_info nvs句柄开启失败 err_code=ESP_ERR_NVS_NOT_FOUND
I (426) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (435) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (444) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (453) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (463) gpio: GPIO[25]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (464) adc_task: calibration scheme version is Line Fitting
I (472) gpio: GPIO[26]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (472) uart: queue free spaces: 10
I (488) mid_res: max_res = 988
I (479) adc_task: Calibration Success
I (495) mid_res: min_res = 231
I (504) BTDM_INIT: BT controller compile version [4012cfb]
I (510) BTDM_INIT: Bluetooth MAC: d8:13:2a:89:32:96
I (516) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (800) main_task: Returned from app_main()
W (800) wifi station: user_wifi_confiG = NULL
[image]

caidamao, 2024-06-28:
以上是我排查的结果，请看看是否有所帮助。如果问题不能解决，我可能需要先回退到1.7.1进行使用。如果需要更多信息，请告诉我。祝好！
The above is the result of my investigation. Please see if it is helpful. If the problem cannot be solved, I may need to rollback to version 1.7.1 for use. If you need more information, please let me know. Wish you all the best!

crazydiamondzgy, 2024-06-30:
yes, the same to me, so i rollback to 1.7.1 ,everything is OK

caidamao, 2024-07-01:
> yes, the same to me, so i rollback to 1.7.1 ,everything is OK

After rolling back, it's indeed normal. I can only look forward to the developers for a fix.

radurentea, 2024-07-01:
Hi @caidamao,

It seems indeed an issue caused by the 1.8.0 release. Can you tell me what is your default terminal in VS Code?

caidamao, 2024-07-01:
> It seems indeed an issue caused by the 1.8.0 release. Can you tell me what is your default terminal in VS Code?

@radurentea cmd

radurentea, 2024-07-01:
Hi @caidamao,

I've created a fix for this. Could you test it by installing the extension using this .vsix file: [esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1654578276.zip) ?

caidamao, 2024-07-01:
> Hi @caidamao,
> 
> I've created a fix for this. Could you test it by installing the extension using this .vsix file: [esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1654578276.zip) ?

安装了“esp-idf-extension.vsix”，情况得到了修复。现在命令输出的时候使用了“”而不是''。我的监视器现在可以正常的打开串口并进行输出了。
I have installed "esp-idf-extension.vsix" and the issue has been resolved. Now the command output is using double quotes ("") instead of single quotes ('') as intended. My monitor can now properly open the serial port and output the expected data.

The fix is effective, that's great!
@radurentea

closed 2024-07-09 by caidamao