espressif/vscode-esp-idf-extension #1232: idf.preBuildTask appends build path before command

isMyNameLuis, 2024-06-28:
### OS

Windows

### Operating System version

Windows 11

### ESP-IDF version

5.2.2

### Python version

3.11.4

### Doctor command output

[report.txt](https://github.com/user-attachments/files/16030072/report.txt)

### Description

I have the preBuildTask defined as bellow:
`"idf.preBuildTask": "${idf.pythonBinPathWin} ${workspaceFolder}/utils/prebuild.py ${workspaceFolder}"`

Unfortunately when building it seems like, since last update, the path of the build directory is pre-appended to the preBuildTask command.

` *  Executing task: c:\Users\luisp\Documents\code\wmini-secc-esp32\build\${idf.pythonBinPathWin} c:\Users\luisp\Documents\code\wmini-secc-esp32\utils\prebuild.py c:\Users\luisp\Documents\code\wmini-secc-esp32  
 *  The terminal process failed to launch: Path to shell executable "c:\Users\luisp\Documents\code\wmini-secc-esp32\build\${idf.pythonBinPathWin} c:\Users\luisp\Documents\code\wmini-secc-esp32\utils\prebuild.py c:\Users\luisp\Documents\code\wmini-secc-esp32" does not exist. `

### Debug Message

```plain
*  Executing task: c:\Users\luisp\Documents\code\wmini-secc-esp32\build\${idf.pythonBinPathWin} c:\Users\luisp\Documents\code\wmini-secc-esp32\utils\prebuild.py c:\Users\luisp\Documents\code\wmini-secc-esp32  

 *  The terminal process failed to launch: Path to shell executable "c:\Users\luisp\Documents\code\wmini-secc-esp32\build\${idf.pythonBinPathWin} c:\Users\luisp\Documents\code\wmini-secc-esp32\utils\prebuild.py c:\Users\luisp\Documents\code\wmini-secc-esp32" does not exist.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

isMyNameLuis, 2024-07-01:
> Hi @isMyNameLuis thanks for the bug reprot. It seems to be an issue with the release version 1.8.0. I will work on a fix today, until it gets fixed, the solution would be to revert the version back to 1.7.1.
> 
> You can do so with this .vsix file: https://github.com/espressif/vscode-esp-idf-extension/releases/download/v1.7.1/esp-idf-extension-1.7.1.vsix
> 
> Release versions can be found here: https://github.com/espressif/vscode-esp-idf-extension/releases

Thank you. Yes in the meantime I've already agreed with the team to rollback while we wait for the fix.

radurentea, 2024-07-04:
Hi @isMyNameLuis,

I've created a fix for this, could you test this version by installing the extension using the following .vsix file? [esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1664025402.zip)

isMyNameLuis, 2024-07-04:
Hello @radurentea ,

Tested and is working perfectly, thank you!

closed 2024-07-25 by github-actions[bot]