espressif/vscode-esp-idf-extension #1233: Monitor Issue after updating the Extension to V1.8.0 On Ubuntu 22.04

parthbhat13, 2024-06-30:
### OS

Linux

### Operating System version

PopOs_Ubuntu22.04

### Visual Studio Code version

1.90.2

### ESP-IDF version

4.4.6

### Python version

3.10.12

### Doctor command output

DevKit

### Extension

```
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: Task ESP-IDF Build exited with code 1\n\tat /dist/extension.js:2:1590016\n\tat n.value (/out/vs/workbench/api/node/extensionHostProcess.js:163:20722)\n\tat l.y (/out/vs/workbench/api/node/extensionHostProcess.js:82:664)\n\tat l.fire (/out/vs/workbench/api/node/extensionHostProcess.js:82:881)\n\tat n.$onDidEndTaskProcess (/out/vs/workbench/api/node/extensionHostProcess.js:156:57296)","level":"error","timestamp":"2024-06-30T18:51:30.019Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /dist/extension.js:2:1464939\n\tat ChildProcess.exithandler (node:child_process:415:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat maybeClose (node:internal/child_process:1105:16)\n\tat Socket.<anonymous> (node:internal/child_process:457:11)\n\tat Socket.emit (node:events:514:28)\n\tat Pipe.<anonymous> (node:net:337:12)","level":"error","timestamp":"2024-06-30T19:01:12.167Z"}
```

### Description

there has been the following issue since the extension got updated to 1.8.0

```
parth7490@dell7490-popos:~/esp32Projects/blink$ & export IDF_PATH='/home/parth7490/esp/v4.4.6/esp-idf'
bash: syntax error near unexpected token `&'
& '/home/parth7490/.espressif/python_env/idf4.4_pyparth7490@dell7490-popos:~/esp32Projects/blink$ & '/home/parth7490/.espressif/python_env/idf4.4_py3.10_env/bin/python' '/home/parth7490/esp/v4.4.6/esp-idf/tools/idf_monitor.py' -p /dev/ttyUSB1 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 '/home/parth7490/esp32Projects/blink/build/blink.elf'
```

ive pasted the problem as shown in the terminal after clicking on the Monitor Device , or Build, Flash, Monitor. 

### Debug Message

```plain
None!!
```

### Other Steps to Reproduce

This seems to be the issue not only on my main laptop, but as well as, seems to be facing it even on my home laptop. mainly after the IDF extension got updated to V1.8.0

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2024-07-01:
Hi @parthbhat13, it seems to be an issue with the release version 1.8.0. I will work on a fix today, until it gets fixed, the solution would be to revert the version back to 1.7.1.

You can do so with this .vsix file: https://github.com/espressif/vscode-esp-idf-extension/releases/download/v1.7.1/esp-idf-extension-1.7.1.vsix

Release versions can be found here: https://github.com/espressif/vscode-esp-idf-extension/releases

parthbhat13, 2024-07-01:
Hi, @radurentea , thanks for the prompt reply, will revert back to V1.7.1 for now then.

radurentea, 2024-07-01:
Hi @parthbhat13,

Can you run "ESP-IDF: Doctor Command"(This command will copy your configuration to the clipboard) from command palette in VS Code? (ctrl + shift + p) and then paste the output here? It will help me reproduce your issue.

I've tried monitoring on linux (fedora) and for me it works. I think doctor command covers it, but can you confirm "bash" is the default terminal in vs code for you?

parthbhat13, 2024-07-01:
Hi @radurentea 

i guess this is what you are looking for? 
```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 6.9.3-76060903-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.local/bin:<HOMEPATH>/.nvm/versions/node/v20.8.0/bin:<HOMEPATH>/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.90.2 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.8.0 

---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.23.1/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.10.2
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf4.4_py3.10_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) board/esp32-wrover-kit-3.3v.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/cmake/3.23.1/bin: false
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/ninja/1.10.2: false
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
```

i just removed the workspace folder name of the file, and i believe what you wanna see mainly is 
```
Visual Studio Code shell /usr/bin/bash 
``` 
this? 
i guess using tabby as my main  terminal would make any difference? i doubt that honestly but we never know...

radurentea, 2024-07-01:
Hi @parthbhat13,

The behaviour is very weird based on the code we've introduced in 1.8.0 release and base on this log from you:
`& export IDF_PATH='/home/parth7490/esp/v4.4.6/esp-idf'`

Can you search the following property`"terminal.integrated.defaultProfile"` in your settings.json file at all 3 levels, using these command palette commands?
[image]

The property can be `"terminal.integrated.defaultProfile.linux"` or if you've had this project on a windows machine or you have a synced Microsoft account, you might have some left overs settings from windows for this property `"terminal.integrated.defaultProfile.windows"`

If you do find any of these properties, can you tell me in which level you've found them (workspace/user/default) ?

radurentea, 2024-07-01:
A work around this, would be to delete this `"terminal.integrated.defaultProfile.windows": "PowerShell"`, because the extension checks this setting in order to take the user's default terminal. 

We will have to think of a way to fix this edge case in the future.

parthbhat13, 2024-07-01:
Hi @radurentea , 

now that did fix the issue. thanks a lot for your time and effort. really appreciate it. 
Cheers..

closed 2024-07-01 by parthbhat13

radurentea, 2024-07-01:
Thank you for the report @parthbhat13, this edge case will be fixed in the next release.