espressif/vscode-esp-idf-extension #1239: Monitor Device button error

mbastida123, 2024-07-03:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.90.1

### ESP-IDF version

4.4.7

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files\WireGuard\;C:\Program Files\Git\cmd;C:\Program Files\Docker\Docker\resources\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python312\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.90.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\cmd.exe 
ESP-IDF Extension version 1.8.0 
Workspace folder c:<HOMEPATH>\Documents\Repos\hmi-fw-esp32 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Users/603mb/esp/v4.4.7/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
ESP-HomeKit-SDK Path (idf.espHomeKitSdkPath) ${env:HOMEKIT_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\esp\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\esp\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\esp\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\esp\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:<HOMEPATH>\esp\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\esp\.espressif\tools\cmake\3.23.1\bin;C:<HOMEPATH>\esp\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:<HOMEPATH>\esp\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\esp\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\esp\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:<HOMEPATH>\esp\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\esp\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
Virtual env Python Path (idf.pythonBinPath) C:/Users/603mb/esp/.espressif/python_env/idf4.4_py3.11_env/Scripts/python.exe
Serial port (idf.port) COM6
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32s3.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\esp\.espressif
Git Path (idf.gitPath) C:/Users/603mb/esp/.espressif/tools/idf-git/2.43.0/cmd/git.exe
Notification Mode (idf.notificationMode) All
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-HomeKit Path (idf.espHomeKitSdkPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\esp\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\esp\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\esp\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\esp\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\esp\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\esp\.espressif\tools\cmake\3.23.1\bin: true
Access to C:<HOMEPATH>\esp\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to C:<HOMEPATH>\esp\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\esp\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\esp\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
[... 137 lines cut ...]
{"message":"Cannot access filePath: ${env:ADF_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\603mb\\AppData\\Local\\Programs\\Microsoft VS Code\\${env:ADF_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (\\dist\\extension.js:2:1461757)\n\tat \\dist\\extension.js:2:1685466\n\tat Fs (\\dist\\extension.js:2:1686534)\n\tat async \\dist\\extension.js:2:1902761","level":"error","timestamp":"2024-07-03T13:53:01.060Z"}
{"message":"Cannot access filePath: ${env:MDF_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\603mb\\AppData\\Local\\Programs\\Microsoft VS Code\\${env:MDF_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (\\dist\\extension.js:2:1461757)\n\tat \\dist\\extension.js:2:1685644\n\tat Fs (\\dist\\extension.js:2:1686534)\n\tat async \\dist\\extension.js:2:1902761","level":"error","timestamp":"2024-07-03T13:53:01.060Z"}
{"message":"Cannot access filePath: ${env:ESP_MATTER_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\603mb\\AppData\\Local\\Programs\\Microsoft VS Code\\${env:ESP_MATTER_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (\\dist\\extension.js:2:1461757)\n\tat \\dist\\extension.js:2:1685736\n\tat Fs (\\dist\\extension.js:2:1686534)\n\tat async \\dist\\extension.js:2:1902761","level":"error","timestamp":"2024-07-03T13:53:01.060Z"}
{"message":"Cannot access filePath: ${env:HOMEKIT_PATH}","stack":"Error: ENOENT: no such file or directory, access 'C:\\Users\\603mb\\AppData\\Local\\Programs\\Microsoft VS Code\\${env:HOMEKIT_PATH}'\n\tat Object.access (node:internal/fs/sync:42:11)\n\tat Object.accessSync (node:fs:248:10)\n\tat t.accessSync (node:electron/js2c/node_init:2:8888)\n\tat Se (\\dist\\extension.js:2:1461757)\n\tat \\dist\\extension.js:2:1685832\n\tat Fs (\\dist\\extension.js:2:1686534)\n\tat async \\dist\\extension.js:2:1902761","level":"error","timestamp":"2024-07-03T13:53:01.061Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1464939\n\tat ChildProcess.exithandler (node:child_process:415:7)\n\tat ChildProcess.emit (node:events:514:28)\n\tat maybeClose (node:internal/child_process:1105:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:305:5)","level":"error","timestamp":"2024-07-03T13:53:02.225Z"}

--------------------------------------------------------------------------------------------------------------------------------------------

### Description

When I click on Monitor Device button located in the lower part I get an error. If I remove the ' in the command the idf extension executes everything works. like this:

`C:/Users/603mb/esp/.espressif/python_env/idf4.4_py3.11_env/Scripts/python.exe C:\Users\603mb\esp\v4.4.7\tools\idf_monitor.py -p COM6 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 c:\Users\603mb\Documents\Repos\hmi-fw-esp32\build\hmi-fw-esp32.elf`

### Debug Message

```plain
[... cut ...] nv/Scripts/python.exe' 'C:\Users\603mb\esp\v4.4.7\tools\idf_monitor.py' -p COM6 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 'c:\Users\603mb\Documents\Repos\hmi-fw-esp32\build\hmi-fw-esp32.elf'
"'C:" no se reconoce como un comando interno o externo,
programa o archivo por lotes ejecutable.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2024-07-04:
This is a duplicate of #1230 and #1232.

Please review this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1656065362.zip) from bug fix PR #1234

closed 2024-07-18 by brianignacio5

Digitalisationtransformers, 2024-11-18:
/
[image]

PR #1234 (2024-07-18): Fix monitoring for Command Prompt
## Description

- Fix monitoring for Command Prompt
- Fix "Git Bash" white space support
- Fix edge-case for users that have default terminal settings from older setups. We now correctly identify the shell that's been used by the user

Fix https://github.com/espressif/vscode-esp-idf-extension/issues/1230
Fix https://github.com/espressif/vscode-esp-idf-extension/issues/1233
Fix #1239 
Fix #1242

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Monitor a flashed project while having "Command Prompt" as  [... cut]